hackbnw/faustus #1: Not working in fx505dy

thor0dinson, 2019-06-22:
I'm trying to install it in my Asus TUF FX505DY
While doing `modprobe faustus` it gives me `modprobe: ERROR: could not insert 'faustus': No such device`
Is the driver only for the FX505GM model ?

DMI info:

        BIOS Information
            Vendor: American Megatrends Inc
            Version: FX505DY.308
            Release Date: 03/19/2019
        Base Board Information
            Manufacturer: ASUSTeK COMPUTER INC.
            Product Name: FX505DY
            Version: 1.0

hackbnw, 2019-06-24:
It was only tested with the *GM version, you can try loading it at your own risk with:
`sudo modprobe faustus let_it_burn=1`

if everything works as expected, just post a reply and I will add your model to DMI check. Thanks!

sheepandshepherd, 2019-06-24:
The DY models lack the RGB keyboard backlight, IIRC.

I can test this with a similar AMD model, FX505DD, in a day or two. It does have the RGB backlight like yours. Anything in particular to look for in the DSDT, or should I just post it here?

ryanobeirne, 2019-06-25:
I have a FX505DY and cannot load this module either. I'm fairly certain this laptop has only a single color (red) keyboard backlight. The keyboard backlight hotkeys do not register at all.

I built the module on ArchLinux 5.1.14-arch1-1-ARCH, then:

    # insmod src/faustus.ko

yields the following error:

    insmod: ERROR: could not insert module src/faustus.ko: No such device

dmesg yields no messages. Then I ran the following:

    # insmod src/faustus.ko let_it_burn=1

I get the same error from insmod, but dmesg says this:

```
[  651.828927] faustus: Omitting DMI verification
[  651.829020] input: Faustus Hotkey Input as /devices/platform/faustus/input/input17
[  651.829219] faustus: Failed to install WMI notify handler: AE_ALREADY_ACQUIRED
[  651.844283] audit: type=1130 audit(1561426375.800:52): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-backlight@leds:faustus::kbd_backlight comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[  651.858653] faustus: probe of faustus failed with error -5
[  651.858672] faustus: Can't probe platform driver: -19
```

At this point, the keyboard backlight hotkeys register errors in dmesg:

```
[  681.756806] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA2] at bit offset/length 64/32 exceeds size of target Buffer (64 bits) (20190215/dsopcode-198)
[  681.756817] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20190215/psparse-529)
[  681.756848] WARNING: CPU: 0 PID: 5 at drivers/leds/led-class.c:151 led_classdev_notify_brightness_hw_changed+0x1f/0x30
[  681.756849] Modules linked in: cmac rfcomm bnep btusb btrtl btbcm btintel bluetooth uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev ecdh_generic media joydev mousedev nls_iso8859_1 amdgpu nls_cp437 vfat fat hid_multitouch hid_generic edac_mce_amd kvm_amd ccp kvm irqbypass 8821ce(OE) crct10dif_pclmul asus_nb_wmi asus_wmi sparse_keymap crc32_pclmul wmi_bmof snd_hda_codec_realtek ghash_clmulni_intel chash snd_hda_codec_generic gpu_sched amd_iommu_v2 ledtrig_audio i2c_algo_bit ttm snd_hda_codec_hdmi drm_kms_helper snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep aesni_intel snd_pcm aes_x86_64 drm input_leds pcspkr snd_timer i2c_hid crypto_simd cfg80211 cryptd tpm_crb glue_helper snd k10temp hid r8169 soundcore sp5100_tco i2c_piix4 realtek agpgart libphy syscopyarea tpm_tis sysfillrect rfkill sysimgblt fb_sys_fops tpm_tis_core tpm ac wmi evdev battery asus_wireless rng_core pinctrl_amd mac_hid pcc_cpufreq acpi_cpufreq crypto_user ip_tables x_tables ext4
[  681.756914]  crc32c_generic crc16 mbcache jbd2 sd_mod serio_raw atkbd libps2 ahci libahci crc32c_intel libata xhci_pci scsi_mod xhci_hcd i8042 serio
[  681.756928] CPU: 0 PID: 5 Comm: kworker/0:0 Tainted: G           OE     5.1.14-arch1-1-ARCH #1
[  681.756930] Hardware name: ASUSTeK COMPUTER INC. TUF Gaming FX505DY_FX505DY/FX505DY, BIOS FX505DY.304 01/17/2019
[  681.756935] Workqueue: kacpi_notify acpi_os_execute_deferred
[  681.756939] RIP: 0010:led_classdev_notify_brightness_hw_changed+0x1f/0x30
[  681.756942] Code: 44 89 c0 c3 0f 1f 80 00 00 00 00 0f 1f 44 00 00 4c 8b 87 38 01 00 00 4d 85 c0 74 0e 89 b7 34 01 00 00 4c 89 c7 e9 81 61 c5 ff <0f> 0b c3 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 0f 1f 44 00 00
[  681.756943] RSP: 0018:ffffa530c320bda0 EFLAGS: 00010246
[  681.756946] RAX: 00000000fffffffb RBX: ffff9d42470dc800 RCX: 0000000000000010
[  681.756947] RDX: ffffffffa6a8bb10 RSI: 0000000000000000 RDI: ffff9d42470dcaf8
[  681.756948] RBP: ffff9d419dc05f60 R08: 0000000000000000 R09: ffffffffa61283e0
[  681.756949] R10: ffff9d4256807600 R11: 0000000000000001 R12: ffff9d42470dcaf8
[  681.756950] R13: ffff9d4256c29d00 R14: 0000000000000000 R15: 0ffff9d4256c29d1
[  681.756952] FS:  0000000000000000(0000) GS:ffff9d4256c00000(0000) knlGS:0000000000000000
[  681.756954] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  681.756955] CR2: 000018589ae10000 CR3: 00000008026f8000 CR4: 00000000003406f0
[  681.756956] Call Trace:
[  681.756967]  asus_wmi_notify+0x21b/0x2c0 [asus_wmi]
[  681.756975]  acpi_wmi_notify_handler+0x167/0x170 [wmi]
[  681.756980]  ? finish_task_switch+0x84/0x2e0
[  681.756984]  ? __switch_to_asm+0x35/0x70
[  681.756989]  acpi_ev_notify_dispatch+0x4a/0x5f
[  681.756992]  acpi_os_execute_deferred+0x16/0x20
[  681.756996]  process_one_work+0x1d1/0x3e0
[  681.757000]  worker_thread+0x4a/0x3d0
[  681.757004]  kthread+0xfb/0x130
[  681.757006]  ? process_one_work+0x3e0/0x3e0
[  681.757008]  ? kthread_park+0x90/0x90
[  681.757011]  ret_from_fork+0x22/0x40
[  681.757016] ---[ end trace 558948239af62d74 ]---

```

Here's a link to my dsdt.dsl:

https://www.codepile.net/pile/AOEngGKv

I hope this helps. Please let me know if you need any help testing.

sheepandshepherd, 2019-06-26:
Got similar results on the FX505DD. I put the DSDT, DMI info, and dmesg errors [here](https://github.com/sheepandshepherd/fx505dd-dsdt).

prmsrswt, 2019-06-29:
I also own a FX505DY, and I also go similar results as @ryanobeirne .

The dmesg output after running `sudo insmod src/faustus.ko let_it_burn=1` is
```
[ 6676.939976] faustus: Omitting DMI verification
[ 6676.940123] input: Faustus Hotkey Input as /devices/platform/faustus/input/input18
[ 6676.940912] faustus: Failed to flush the WMI event queue: -5
[ 6676.958013] audit: type=1130 audit(1561830639.446:73): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-backlight@leds:faustus::kbd_backlight comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[ 6677.008926] faustus: probe of faustus failed with error -5
[ 6677.008944] faustus: Can't probe platform driver: -19
```
The output of `sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3`
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505DY.308
        Release Date: 03/19/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505DY
        Version: 1.0
```

dsdt.aml : https://www.dropbox.com/s/7uj2kloioo4j88x/dsdt.aml?dl=0

ryanobeirne, 2019-06-29:
I upgraded the BIOS to [v309](https://dlcdnets.asus.com/pub/ASUS/GamingNB/FX505DY/FX505DYAS309.zip) and got the same results, with this kernel module. However, my dsdt.dsl is slightly different now:

https://www.codepile.net/pile/A47oWg4g

Also, I didn't post this before but here's this:

```
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FX505DY.309
	Release Date: 04/25/2019
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FX505DY
	Version: 1.0       
```

hackbnw, 2019-07-01:
Thanks everyone! There are couple of issues here.

I guess that on FX505DD the issue is that the original asus_wmi / asus_nb_wmi drivers (or something other) did load (surprisingly, mine fails every time, which kernel / distro versions do you have?) and registered on the WMi event on boot (grep for asus or wmi in dmesg). You could try blacklisting them 
https://askubuntu.com/questions/110341/how-to-blacklist-kernel-modules

The issue on FX505DY is a bug (this code does not handle Return (Ones) on event queue end), I've made this driver rather universal in mainline patch version since this one was written, so I'm going to partially backport that asus_wmi driver and replace the one in this repository with it shortly.

sheepandshepherd, 2019-07-01:
Perfect, the original asus_wmi/asus_nb_wmi were indeed being loaded. With both blacklisted, this module can load without issues on the DD model. Thanks!

Backlight brightness, RGB, and fan mode can all be set properly using the device files. Key events all work except for the Fn-F5 fan mode key, which `evtest` recognizes as KEY_UNKNOWN with value `ae` instead of `99` (output is [here](https://github.com/sheepandshepherd/fx505dd-dsdt/blob/master/evtest.log)).

Rest of the OS/kernel info:

|Model  |BIOS  |OS  |Kernel version |
|- |- |- |- |
|FX505DD |FX505DD.305 |OpenSUSE Tumbleweed |5.1.7-1-default |

prmsrswt, 2019-07-02:
@hackbnw thanks for looking into this. In ASUS FX505DY the kernel modules `asus_wmi` and `asus_nb_wmi` loads at boot, but `asus_nb_wmi` fails. Here's the relevent dmesg
```
dmesg | grep asus
[    3.321903] asus_wmi: ASUS WMI generic driver loaded
[    3.357084] input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input10
[    3.365759] asus_wmi: Number of fans: 0
[    3.495297] asus-nb-wmi: probe of asus-nb-wmi failed with error -5
```

I am running Arch Linux with kernel `5.1.15-arch1-1-ARCH` and BIOS `FX505DY.308`

ryanobeirne, 2019-07-02:
I'm also running Arch and I get the same results as @prmsrswt

hackbnw, 2019-07-04:
Ok, so I've hacked together the backport of patched driver and added both FX505DD and FX505DY to DMI check, you can try if it works now. Also, README is updated, please check the hints on configuration there.

S-trace, 2019-07-04:
Please also add FX705GE:
DMI_MATCH(DMI_SYS_VENDOR, "ASUSTeK COMPUTER INC."),
DMI_MATCH(DMI_PRODUCT_NAME, "TUF Gaming FX705GE_FX705GE"),

prmsrswt, 2019-07-04:
Great work @hackbnw but I the module does not load for me. I have blacklisted both `asus_wmi` and `asus_nb_wmi`.
It fails with following error message after running `sudo insmod src/faustus.ko`
```
insmod: ERROR: could not insert module src/faustus.ko: Unknown symbol in module
```
`dmesg` output
```
[   92.789800] faustus: Unknown symbol sparse_keymap_report_event (err -2)
[   92.789851] faustus: Unknown symbol sparse_keymap_setup (err -2)
```

I am running Arch Linux with kernel `5.1.15-arch1-1-ARCH`, and I am currently on BIOS version 308 (v309 is latest).

hackbnw, 2019-07-04:
My bad, thanks! Please run
```
sudo modprobe sparse-keymap wmi video
```
this is only required before inserting using insmod. I've updated README.

hackbnw, 2019-07-04:
@S-trace does it work for you on FX705GE?

prmsrswt, 2019-07-04:
> My bad, thanks! Please run
> 
> ```
> sudo modprobe sparse-keymap wmi video
> ```
> 
> this is only required before inserting using insmod. I've updated README.

Thanks, I just figured that out. `sparse_keymap` was not loaded in my case.

Keyboard backlight and Touchpad Fn key works like a charm. The `dmesg` output was
```
[ 1454.318397] faustus: DMI checK: FX505DY
[ 1454.318918] faustus: Initialization: 0x1
[ 1454.319069] faustus: BIOS WMI version: 9.0
[ 1454.319168] faustus: SFUN value: 0x21
[ 1454.319173] faustus faustus: Use DSTS
[ 1454.319175] faustus faustus: Enable event queue
[ 1454.322623] input: Asus WMI hotkeys as /devices/platform/faustus/input/input17
[ 1454.322963] faustus: Number of fans: 0
```
Number of fans detected is 0 so Fan control is not working but that's not a major issue for me.
Thanks for your hard work.

S-trace, 2019-07-04:
@hackbnw Yes, it works - keyboard backlight setup, touchpad toggle, keyboard backlight control using hotkeys. 
But fan speed can't be set or read.

Did you received my email i sent yesterday?

closed 2019-07-07 by hackbnw

Shadefix, 2021-12-25:
not sure if this thread is still relevant but i have an Asus tuf FX505DY.316 model.
dmsg | tail spits out 

[  381.823317] faustus: DMI check: FX505DY
[  381.823476] faustus: Initialization: 0x1
[  381.823523] faustus: BIOS WMI version: 9.0
[  381.823555] faustus: SFUN value: 0x21
[  381.823557] faustus faustus: Use DSTS
[  381.823558] faustus faustus: Enable event queue
[  381.830463] input: Asus WMI hotkeys as /devices/platform/faustus/input/input17
[  381.834892] faustus faustus: Led asus::kbd_backlight renamed to asus::kbd_backlight_1 due to name collision
[  381.836532] faustus: Unable to register notify handler - 21
[  381.917386] faustus: Can't probe platform driver: -19

$blacklist asus_wmi
$blacklist asus_nb_wmi
these work since the backlight stays off on boot. standard its on rainbow breathing.

i have no idea what to do, i tried making sense of this thread but i am not that experienced with linux.

junagu, 2023-04-04:
I don't know if anyone is still maintaining this. I'm trying to install this on a TUF 505 but when `make` in the faustus folder I get:

`make -C /lib/modules/6.1.0-6mx-amd64/build M=/home/****/opt/faustus-master modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-6mx-amd64'
  CC [M]  /home/****/opt/faustus-master/src/faustus.o
/home/****/opt/faustus-master/src/faustus.c: In function âasus_wmi_addâ:
/home/****/opt/faustus-master/src/faustus.c:3111:3: error: implicit declaration of function âacpi_video_set_dmi_backlight_typeâ; did you mean âacpi_video_get_backlight_typeâ? [-Werror=implicit-function-declaration]
 3111 |   acpi_video_set_dmi_backlight_type(acpi_backlight_vendor);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   acpi_video_get_backlight_type
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-6mx-common/scripts/Makefile.build:255: /home/****/opt/faustus-master/src/faustus.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-6mx-common/Makefile:2030: /home/****/opt/faustus-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-6mx-amd64'
make: *** [Makefile:9: default] Error 2
`

and thus when trying `sudo insmod src/faustus.ko`
it gives me :
`insmod: ERROR: could not load module ./src/faustus.ko: No such file or directory`

cromerc, 2023-04-04:
> I don't know if anyone is still maintaining this. I'm trying to install this on a TUF 505 but when `make` in the faustus folder I get:
>
> `make -C /lib/modules/6.1.0-6mx-amd64/build [...]

Kernel 6.1 already has this stuff built in. You don't need this.

cromerc, 2023-04-04:
There isn't really anything to configure. These models of notebook only have 3 fan speeds. Silent, balanced, and turbo. Other than that, there isn't any control beyond that.

If its constantly spinning up and stopping, there are 2 posibilites. Faulty/dirty hardware,or the temperature is fluctuating between the threshold needed to change fan speeds. If the notebook is older, try cleaning the fans and removing dust.