hackbnw/faustus #3: Unknown key ae pressed when pressing fan button on FX505DY

Brigapes, 2019-07-26:
Installed sucessfully, backlight works, non-rgb keyboard variant, on 19.04 Ubuntu, fresh install.
Getting  "Unknown key ae pressed when pressing fan button on FX505DY" on pressing FN+F5, any other info i can provide just let me know.

Djip007, 2019-08-31:
Same her with FX505DY and Fedora30...  FN+F5 == FAN_BOOST => ae KEY not 99
And the dev_id is 0x00120075 not 0x00110018 like define. (look some dsdt have both... not this one.)

I test making change in kernel 5.3_RC6 with 
#define NOTIFY_KBD_FBM                 0xae
#define ASUS_WMI_DEVID_FAN_BOOST_MODE  0x00120075 

CPU fan ctrl work ( low, normal & boost)...

look like 0x00110013 & 0x00110014 report the 2 FAN speeds... (dont know factor to apply... )

If I can help...

[dsdt-FX505DY-311.dsl.txt](https://github.com/hackbnw/faustus/files/3561959/dsdt-FX505DY-311.dsl.txt)

Djip007, 2019-08-31:
not sur but:
0x00110022 and 0x00110023 report 0x100ff... 
and may change FAN speed (FF==AUTO? ... )
But did not test... DSDT have code related with  0x00120075 (FAN auto control low/normal/boost...)

Djip007, 2019-09-13:
up...

hackbnw, 2019-11-13:
Ok, so I've added support for both 0xAE as switch key and the 0x00120075 device ID. 

Can't really test the 0x00110013 and 0x00110014, but it certainly looks like it judging by RTAH there. Can it be revolutions per minute? 

About DSDT database, not that I know of. You can find some in other issues here and on the internet, otherwise some people on platform-driver-x86 kernel list may help if you write a patch. On my device both 0x00110013 and 0x00110014 are empty and return 0 in DSTS and 1 in DEVS, the 0x00110022 and 0x00110023 are not there at all.

closed 2019-11-13 by hackbnw

Djip007, 2019-12-14:
look nice! thanks/