hackbnw/faustus #9: Infinite Loop Setting Fan Boost Mode

duane-r, 2020-01-04:
When I set fan boost mode directly with
`echo "2" > /sys/devices/platform/faustus/fan_mode`
the driver goes into an infinite loop, calling fan_mode_store repeatedly. Is the return value from fan_mode_store correct? If I return (ssize_t)2, the problem seems to go away.

I'm having a hard time following the code, not being much of a C programmer.

S-trace, 2020-01-05:
Hello.

According to [kernel's documentation for sysfs](https://www.kernel.org/doc/Documentation/filesystems/sysfs.txt):
```
- store() should return the number of bytes used from the buffer. If the
  entire buffer has been used, just return the count argument.
```
So, it should return (ssize_t) 2 for echo "2" > /sys/devices/platform/faustus/fan_mode (actually, echo append a '\n' to its argument, and really write '2\n' to the sysfs file, which is 2 bytes).
But please don't hardcode this (ssize_t) 2 - user may write '255\n' to the sysfs file or so.

closed 2020-01-18 by hackbnw

hackbnw, 2020-01-18:
Should work now, thanks to @S-trace for the fix!