hackbnw/faustus #10: Works on the ASUS FX505DT laptop (AMD version)

decio-fonini, 2020-01-11:
I'm happy to report that faustus works flawlessly on the ASUS TUF FX505DT:  all options from set_rgb.sh work, as does the fan control. Thank you!

------------------------------------------------
CPU information:

[]# lscpu | grep Ryzen
Model name:      AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
[]# 

------------------------------------------------
Board information:

[]# dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505DT.304
        Release Date: 05/02/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505DT
        Version: 1.0
[]# 
------------------------------------------------
Attached: DSDT

[FX505DT-AMD-dsdt.aml.txt](https://github.com/hackbnw/faustus/files/4049465/FX505DT-AMD-dsdt.aml.txt)

luiq54, 2020-01-14:
Seconded. Works amazing on my Tuf 705DT.

System:
Ryzen 5-3550H
Arch Linux - 5.4.11

> $ sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3 

BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX705DT.308
        Release Date: 09/19/2019
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX705DT
        Version: 1.0`

Attached DSDT
[dsdt.aml.zip](https://github.com/hackbnw/faustus/files/4060548/dsdt.aml.zip)

hackbnw, 2020-01-18:
Thanks for your reports! Both FX505DT and FX705DT are now included.

closed 2020-01-18 by hackbnw

icodelifee, 2020-01-19:
> Seconded. Works amazing on my Tuf 705DT.
>
> System:
> Ryzen 5-3550H
> Arch Linux - 5.4.11
>
>
> BIOS Information
> Vendor: American Megatrends Inc.
> Version: FX705DT.308
> Release Date: 09/19/2019
> Base Board [...]

isnt 705DT Ryzen 7 3750H?

Corrupter-rot, 2021-07-27:
@decio-fonini @luiq54 A quick question as I am really concerned.

I only get around 2 hours of battery life on my laptop for medium usage and multitasking & 1.5 hr when heavy use. What your battery backup?

commit d068a2b9 (2020-01-18): Include FX505DT and FX705DT in supported model list, closes #10