hackbnw/faustus #12: Can't run make dkms

luiq54, 2020-01-19:
I am running Arch Linux (Kernel 5.4.12) on Tuf FX705DT

On running `sudo make dkms` , I get
```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j8 KERNELRELEASE=5.4.12-arch1-1 -C /usr/lib/modules/5.4.12-arch1-1/build SUBDIRS=/var/lib/dkms/faustus/0.2/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.12-arch1-1 (x86_64)
Consult /var/lib/dkms/faustus/0.2/build/make.log for more information.
```

Log file `/var/lib/dkms/faustus/0.2/build/make.log` : 
```
DKMS make.log for faustus-0.2 for kernel 5.4.12-arch1-1 (x86_64)
Sun Jan 19 12:54:57 IST 2020
make: Entering directory '/usr/lib/modules/5.4.12-arch1-1/build'
make[1]: Nothing to be done for 'objtool'.
scripts/Makefile.build:42: arch/x86/entry/syscalls/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/x86/entry/syscalls/Makefile'.  Stop.
make: *** [arch/x86/Makefile:238: archheaders] Error 2
make: *** Waiting for unfinished jobs....
scripts/Makefile.modbuiltin:26: usr/Makefile: No such file or directory
make[1]: *** No rule to make target 'usr/Makefile'.  Stop.
make: *** [Makefile:1300: _modbuiltin_usr] Error 2
scripts/Makefile.modbuiltin:26: init/Makefile: No such file or directory
make[1]: *** No rule to make target 'init/Makefile'.  Stop.
make: *** [Makefile:1300: _modbuiltin_init] Error 2
scripts/Makefile.modbuiltin:26: certs/Makefile: No such file or directory
make[1]: *** No rule to make target 'certs/Makefile'.  Stop.
make: *** [Makefile:1300: _modbuiltin_certs] Error 2
scripts/Makefile.modbuiltin:26: kernel/dma/Makefile: No such file or directory
scripts/Makefile.modbuiltin:26: kernel/cgroup/Makefile: No such file or directory
make[2]: *** No rule to make target 'kernel/dma/Makefile'.  Stop.
make[2]: *** No rule to make target 'kernel/cgroup/Makefile'.  Stop.
make[1]: *** [scripts/Makefile.modbuiltin:55: kernel/cgroup] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.modbuiltin:55: kernel/dma] Error 2
scripts/Makefile.modbuiltin:26: kernel/events/Makefile: No such file or directory
make[2]: *** No rule to make target 'kernel/events/Makefile'.  Stop.
make[1]: *** [scripts/Makefile.modbuiltin:55: kernel/events] Error 2
scripts/Makefile.modbuiltin:26: kernel/bpf/Makefile: No such file or directory
make[2]: *** No rule to make target 'kernel/bpf/Makefile'.  Stop.
scripts/Makefile.modbuiltin:26: kernel/livepatch/Makefile: No such file or directory
make[2]: *** No rule to make target 'kernel/livepatch/Makefile'.  Stop.
make[1]: *** [scripts/Makefile.modbuiltin:55: kernel/bpf] Error 2
make[1]: *** [scripts/Makefile.modbuiltin:55: kernel/livepatch] Error 2
scripts/Makefile.modbuiltin:26: kernel/irq/Makefile: No such file or directory
make[2]: *** No rule to make target 'kernel/irq/Makefile'.  Stop.
make[1]: *** [scripts/Makefile.modbuiltin:55: kernel/irq] Error 2
make: *** [Makefile:1300: _modbuiltin_kernel] Error 2
scripts/Kbuild.include:120: *** Recursive variable 'CC_OPTION_CFLAGS' references itself (eventually).  Stop.
make: *** [Makefile:1300: _modbuiltin_arch/x86] Error 2
make: Leaving directory '/usr/lib/modules/5.4.12-arch1-1/build'
```

ryanobeirne, 2020-01-19:
I get the same thing on this kernel on Arch on the FX505DY.

MagicRB, 2020-01-20:
The same thing happens on my friend's laptop on void. (I know some people hate the "me too" type of comments, but I felt it necessary since I run a pretty exotic distro)

Saancreed, 2020-01-22:
A patch like this fixed this issue for me:
```
diff --git a/dkms.conf b/dkms.conf
index 4b9ca4e..0b67d09 100644
--- a/dkms.conf
+++ b/dkms.conf
@@ -1,12 +1,11 @@
 PACKAGE_VERSION="0.2"
 
 PACKAGE_NAME="faustus"
-MAKE[0]="make -C ${kernel_source_dir} SUBDIRS=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build modules"
-CLEAN="make -C ${kernel_source_dir} SUBDIRS=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build clean"
+MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build modules"
 
 BUILT_MODULE_NAME[0]="faustus"
 BUILT_MODULE_LOCATION[0]="src/"
-DEST_MODULE_LOCATION[0]="/extra"
+DEST_MODULE_LOCATION[0]="/kernel/drivers/platform/x86"
 
 REMAKE_INITRD="yes"
 AUTOINSTALL=yes
```

k1-801, 2020-01-26:
Simply removing both MAKE lines works too.

luiq54, 2020-03-09:
@Saancreed Yep that fixed the issue. Thanks!

I will close this issue as soon as fix is in the master.

UtkarshVerma, 2020-04-05:
Does #23 look similar?

hackbnw, 2020-05-02:
@UtkarshVerma I think it is duplicate.

Thanks for the patch @Saancreed ! I've commited the one from #23 onto master, as `DEST_MODULE_LOCATION` should not actually play a role there and I would like to keep clean command for good measure. Please say if that didn't fixed it.

UtkarshVerma, 2020-05-03:
@hackbnw I've installed the latest commit on my laptop and seems to work fine as shown by `dmesg`.

closed 2020-05-04 by hackbnw

UtkarshVerma, 2020-05-05:
@hackbnw After reboot, `faustus` doesn't log anything about the fan speed. Nothing shows up in `dmesg` no matter how many keypresses I make. On the other hand, RGB coloring works fine using OpenRGB.
Do you think this bug is related to this issue or should I post a new one?

> I'm using on the latest commit as of now.