hackbnw/faustus #13: Please Fix Display Flicker Issue On FX505DT

icodelifee, 2020-01-19:
Hey, Ive Installed faustus mod on arch linux and the features you have listed in this module is working fine.
Im opening this issue because its not this modules pecific issue with a issue with linux in this laptop.
So in brief im experiencing display flicker/blinking issue with my FX505DT on litreally all the distros, so can you please look into it and provide us a fix if you can?

Model : FX505DT with Ryzen 5 3550H and GTX1650 
Display Is 120Hz

ryanobeirne, 2020-01-19:
This is not the place for these kinds of issues. That said, you should try setting your display refresh rate to 60Hz.

icodelifee, 2020-01-19:
> 
> 
> This is not the place for these kinds of issues. That said, you should try setting your display refresh rate to 60Hz.

Display Doesnt Support "60Hz"
if it goes down 120 it starts flickering

Tvaroh00, 2020-01-20:
Display flicker isn't related to this driver, though if the maintainer doesn't object, I'd suggest to make this issue the place to solve the flicker, since there is no other real place where we can have this discussion. So, it only happens after I switch to 60, and once I do, it appears even if I if switch back to 120. I will share my modelines, please do the same.

```
# 1920x1080 60.00 Hz
Modeline "1920x1080_60.00"  335.470 1920 1968 2000 2180  1080 1083 1088 2564 +hsync -vsync
# 1920x1080 90.00 Hz
Modeline "1920x1080_90.00"  335.470 1920 1968 2000 2180  1080 1083 1088 1923 +hsync -vsync
# 1920x1080 120.00 Hz
Modeline "1920x1080_120.00"  335.470 1920 1968 2000 2180  1080 1083 1088 1282 +hsync -vsync
```

icodelifee, 2020-01-20:
> Display flicker isn't related to this driver, though if the maintainer doesn't object, I'd suggest to make this issue the place to solve the flicker, since there is no other real place where we can [...]

hey, these are custom mode lines right?
cause if once i tried switching to 60 using xrandr i got a full glitched screen

hackbnw, 2021-03-14:
@icodelifee the issue is very likely not related to this driver, but more likely to GPU one. Can't say for sure, but if it's the same with or without it...

More useful suggestions, did you try the proprietary nvidia drivers (I guess in Arch it's nvidia package) or are you using the open source ones which come with the default kernel (nouveau)? Did you try nouveau.modeset=0 kernel boot parameter? Did you try following this guide https://wiki.archlinux.org/index.php/NVIDIA ? The graphics is always pain.

Didn't see no blinking, but I do occasionally get glitched textures in some GPU accelerated applications with proprietary nvidia driver after waking up from sleep, but nothing serious.

icodelifee, 2021-03-14:
> @icodelifee the issue is very likely not related to this driver, but more likely to GPU one. Can't say for sure, but if it's the same with or without it...
>
> More useful suggestions, did you try the [...]

oh yea, I've tried almost all possible configurable solution but none worked, ended up going back windows now.
according to my understanding, it happens when the screen turns off and the clock speeds get messed up when it turns on back.
The temporary fix I found was to just force off dpms using
`xset dpms force off`
and waking up the display manually.
anyway I've ended up with a conclusion that this device isn't meant for using Linux or at least for me
fixing random bugs popping up everywhere and me trying to patch it just wastes my time, when I could just boot to windows and use the machine just fine.
anyway thanks for responding.
- icodelife

hackbnw, 2021-03-14:
> anyway I've ended up with a conclusion that this device isn't meant for using Linux or at least for me
> fixing random bugs popping up everywhere and me trying to patch it just wastes my time, when I could just boot to windows and use the machine just fine.

Now where would be the fun in that :smile: I think though it is a big part because of Arch. Did run it some time ago, don't get me wrong, it's cool, especially regarding running latest versions, but you should be motivated enough to tinker with it. Now one can say many things about Ubuntu too, but it mostly works.

hackbnw, 2021-03-14:
Thanks for posting the solution! I think it is useful in case anyone else would search for it.

closed 2021-03-14 by hackbnw

hackbnw, 2021-06-02:
Just bumped into this note https://github.com/tromoto/faustus/blob/master/NOTES.txt if anyone else is experiencing the issue