hackbnw/faustus #16: Arch Linux, ASUS TUF FX505DU: RGB confirmed working, but DKMS fails to build

k1-801, 2020-01-26:
I have a laptop of the same series that is not listed in the table present in README.md (ASUS TUF FX505DU). I was looking for a way to manage the keyboard backlight on it. It is not listed as a USB device and it cannot be found by OpenRGB, so it is not i2c/SMBus either, so I attempted to use this particular driver.
There is an AUR package (faustus-dkms-git) that installs this driver, taking the sources directly from this repo.
When trying to install it as-is, I found out that it cannot be build as a DKMS, failing somewhere deep in the Linux code (https://github.com/torvalds/linux/blob/master/scripts/Kbuild.include, line 120, CC_OPTION_CFLAGS refers to itself recursively). Removing the MAKE lines from dkms.conf made it build normally, but it fails to be inserted with `insmod: ERROR: could not insert module src/faustus.ko: No such device`. let_it_burn made it work, set_rgb.sh seems to work.
Please add a full checklist of all actions that need to be performed to confirm the driver 100% functional on a device, and possibly add the device to the supported list.
BIOS version: FX505DU.304, distro Arch Linux, kernel version 5.4.13-arch1-1.

k1-801, 2020-02-07:
New info: seems to cause recursive boot fails, suspend frequently fails to resume or just takes a very long time (over a minute) on FX505DU.304. Sometimes causes one CPU core (CPU#5) to freeze on boot and it doesn't come online until rebooted with no driver loaded.
RGB keyboard seems to work, but with the brightness issue in static mode.
Fan control works both with and without the driver, can't tell if there's any difference.

yaliv, 2020-02-08:
@k1-801 as you know, I use BIOS version 308. That might be case.
As I update the BIOS of my TUF laptop several times, I see that Asus always keeps the last 2 versions in their support site:
https://www.asus.com/Laptops/ASUS-TUF-Gaming-FX505DD-DT-DU/HelpDesk_BIOS/
And I can say that this BIOS has the easiest update method I've ever experience. You don't need a USB drive or a Windows to flash the BIOS. You just need a FAT32 formatted partition (e.g. used for `/boot/efi`). Download the BIOS, extract, copy it to the partition (e.g. call `sudo cp FX505DU-AS.308 /boot/efi`). Then boot to BIOS and you can flash the update in the Advanced Mode. Just find the file in the partition and the flashing process will start.

Next, I'll you what I did and didn't.

I didn't:
- install package faustus-dkms-git (in AUR).

k1-801, 2020-02-09:
I'm going to update BIOS to the latest version (310) and will report shortly.

yaliv, 2020-02-28:
> There is also a possibility that my DKMS build was built incorrectly after all. Applying the paatch mentioned in #12 just in case. Will report if it works this way later.

Hi @k1-801, I modified file `dkms.conf` like in https://github.com/hackbnw/faustus/issues/12#issuecomment-577347785 and now the faustus DKMS module is installed :tada:
```
$ dkms status
faustus, 0.2, 5.5.6-1-MANJARO, x86_64: installed
```

Then I made it load on boot and yes, the driver runs flawlessly :grin:

yaliv, 2020-02-28:
> UPD 2: ended up with an unresponsive black screen again after about a dozen suspends, suspecting conflicts between faustus and the proprietary nvidia driver. It was stuck like this for at least 4 hours.

Hmmm :thinking: about the Nvidia driver, I use the proprietary one.
```
$ __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia inxi -G
Graphics:  Device-1: NVIDIA TU116M [GeForce GTX 1660 Ti Mobile] driver: nvidia v: 440.59
           Device-2: AMD Picasso driver: amdgpu v: kernel
           Display: x11 server: X.Org 1.20.7 driver: amdgpu,nvidia resolution: 1920x1080~120Hz
           OpenGL: renderer: GeForce GTX 1660 Ti/PCIe/SSE2 v: 4.6.0 NVIDIA 440.59
```

As you can see, this driver version has the [PRIME Render Offload](http://download.nvidia.com/XFree86/Linux-x86_64/440.59/README/primerenderoffload.html) feature enabled by default, so the primary GPU is AMD iGPU.
```
$ inxi -G
Graphics:  Device-1: NVIDIA TU116M [GeForce GTX 1660 Ti Mobile] driver: nvidia v: 440.59
           Device-2: AMD Picasso driver: amdgpu v: kernel
           Display: x11 server: X.Org 1.20.7 driver: amdgpu,nvidia resolution: 1920x1080~120Hz
           OpenGL: renderer: AMD RAVEN (DRM 3.36.0 5.5.6-1-MANJARO LLVM 9.0.1) v: 4.5 Mesa 19.3.4
```

---------------------------

Well then.
You said that you got a black screen frequently. Or you got something like a discolored screen.

I do get those screen errors, but not too often. One thing to note is that they happen even when I don't have faustus DKMS module installed.
Perhaps the problem is with the kernel version :man_shrugging:
I used kernel 5.4 before and today I upgraded to 5.5.

I don't have some heavy programs loaded at boot, so the boot time is often quick.
Sometimes I got a black screen or a discolored screen (after SDDM), so the best thing I can do is forcing my laptop to power off.

When resuming from sleep, it always takes a long time but at least the session is fine. I can type my password and continue my work.

By the way, I have this value in `/etc/default/grub`:
```bash
GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_backlight=none apparmor=1 security=apparmor resume=UUID=fc6ff57d-67d0-48e8-b348-a4fd3eae0c29 udev.log_priority=3"
```
The `resume` parameter points to my swap partition. It is particularly useful for resuming from hibernate. Possibly useful for resuming from sleep too? Who knows.

yaliv, 2020-02-29:
Wow, resuming from sleep seems faster now with kernel 5.5 :tada:

yaliv, 2020-02-29:
Wow, resuming from sleep seems quick now with kernel 5.5 :tada:

k1-801, 2020-03-03:
Seems working fine for me too now after several updates.
Had the laptop stuck with 100% cpu load right after resuming a couple of times, but otherwise it works just fine. I guess we can call it working on FX505DU now.

closed 2020-03-03 by k1-801