hackbnw/faustus #17: FX505DV, Ubuntu 19.10 (kernel 5.3.0), RGB control works, fan control doesn't

x0wllaar, 2020-01-29:
The driver does not load with default settings because my model is not supported (as of now). When loaded with let_it_burn=1, it loads.

Keyboard backlight control works without an issue (I might test it more, for now I have tested cycle and solid color modes).

However, when I'm trying to change the fan mode by writing to /sys/devices/platform/faustus/fan_mode, the program (bash, if I try this with `#echo 1 > /sys/devices/platform/faustus/fan_mode`).

I would love if this was resolved and I'm ready to provide any help with this I can.

Thanks in advance.

Distro information:
```
$lsb_release -a                                                                                                       
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan
$uname -a
Linux gregory-laptop 5.3.0-29-generic #31-Ubuntu SMP Fri Jan 17 17:27:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Laptop information:
```
$sudo dmidecode
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.1 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.
Table at 0x000EACC0.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505DV.303
        Release Date: 09/19/2019
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                EDD is supported
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 5.14
        Firmware Revision: 11.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF Gaming FX505DV_FX505DV
        Version: 1.0
        Serial Number: XXXXXXXXXXXXXXXXX
        UUID: XXXXXXXXXXXXXXXXXXXXXXXXXXXXX
        Wake-up Type: Power Switch
        SKU Number:  
        Family: TUF Gaming

```

Driver log (with let_it_burn=1, does not load otherwise):
```
$dmesg | grep faustus
[ 1355.167400] faustus: Omitting DMI verification
[ 1355.167719] faustus: Initialization: 0x1
[ 1355.167868] faustus: BIOS WMI version: 8.1
[ 1355.167990] faustus: SFUN value: 0x4a0061
[ 1355.167994] faustus faustus: Use DSTS
[ 1355.167996] faustus faustus: Enable event queue
[ 1355.175013] input: Asus WMI hotkeys as /devices/platform/faustus/input/input29
[ 1355.175761] faustus: Number of fans: 1
```

hackbnw, 2020-05-02:
There was a bug with writing fan mode via bash (#9). Please pull and see if it persists. It is correct although misleading that it detects only 1 fan. You can only switch between three modes as of now, I am not aware about how and if it is possible to do direct fan control. If it's not the answer, describe what is the issue you have.

hackbnw, 2020-05-04:
Another issue, notice which modes you have when switching with Fn-F5 and observing value of the sysfs files (see readme for fan mode). There might be some laptops where not all of the modes (0, 1, 2) are supported for `fan_boost_mode`.