hackbnw/faustus #18: GUI App For Faustus Module

icodelifee, 2020-02-05:
Since this module works flawlessly for me i thought why not make a GUI app for controlling features such as fan speed and keyboard lighting, well i made one now.
I present you TUF Control, GUI App made for this module for hassle free control of device features
its a WIP and opensource, so PRs are welcome and its written in electron.
[image]

k1-801, 2020-02-12:
OpenRGB also supports keyboard color manipulation, and they are planning on adding fan support.

hackbnw, 2020-05-02:
Wow, that's really cool @icodelifee ! I will add a reference in the README.

Please note that there was rather intense discussion on how to implement RGB keyboard backlight control in platform drivers, when I submitted the patch from this repo in the kernel mailing list. The sysfs API here is therefore still WIP, overall intention was to incorporate it with the new multicolor framework in LED subsystem somehow, but that's a TODO at the moment.

closed 2020-05-02 by hackbnw