hackbnw/faustus #22: Works on FX505DD (but lost fan speed indicator)

Siorki, 2020-03-29:
Running on Debian Bullseye (testing branch)

```
# dmidecode | grep "BIOS Inf\|Board Inf" -A 3 
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505DD.305
        Release Date: 05/06/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505DD
        Version: 1.0

# uname -a
Linux cadmium 5.4.19 #1 SMP Fri Mar 20 22:15:54 CET 2020 x86_64 GNU/Linux

# cat /etc/debian_version 
bullseye/sid
```

Keyboard backlight control works perfectly.

Fan speed control logged  (seen in logs - `dmesg`) but no way to check. I was using the package [psensor](https://packages.debian.org/bullseye/psensor) (GUI over [lm-sensors](https://packages.debian.org/bullseye/lm-sensors) ) to monitor CPU temperature and fan speed, and after disabling both `asus_wmi` and `asus_nb_wmi` the fan speed entry is gone.

hackbnw, 2020-05-02:
Yes, I guess this was fixed in this [patch](https://github.com/torvalds/linux/commit/e3168b874321d04c160c9eb937919eb926ae232f#diff-a746f90c9503689310ee8f3bdc77bc4e) that came out later. I'm going to backport the changes.

hackbnw, 2020-05-02:
Ok, I've merged current linux mainline master branch into the master here, can you check if it works now? I get 0 RPM when running `sensors` with either 5.3 default or the faustus module on FX505GM.

Siorki, 2020-05-16:
Updated to https://github.com/hackbnw/faustus/commit/08f44119b93710d9c3137f6cd52e064ea7b15b03

Also updated the kernel :
```sh
# uname -a
Linux cadmium 5.6.0-1-amd64 #1 SMP Debian 5.6.7-1 (2020-04-29) x86_64 GNU/Linux
```

``fan_speed`` is back in ``sensors``, but always reads 0 RPM.

Siorki, 2020-07-03:
I got correct readings for ``fan_speed`` after rebooting.

On the first time tuning the laptop on, it always reads 0, but after a reboot the value is correctly set.
Maybe this has something to do with the computer booting too fast for the sensor to initialize in time.

chknatwork, 2021-04-13:
This happens to me as well in a FX505DR-HN657T.