hackbnw/faustus #23: `make` fails on linux-5.6.1 kernel

UtkarshVerma, 2020-04-04:
I have ASUS FX505DT. My current linux kernel is `5.6.1-050601-generic`.
When I run the `make` command, this is what I get:
```
$ make
make -C /lib/modules/5.6.1-050601-generic/build M=/home/utkarsh/GitHub/Others/faustus modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.1-050601-generic'
  CC [M]  /home/utkarsh/GitHub/Others/faustus/src/faustus.o
/home/utkarsh/GitHub/Others/faustus/src/faustus.c: In function ‘asus_hwmon_temp1’:
/home/utkarsh/GitHub/Others/faustus/src/faustus.c:1698:10: error: implicit declaration of function ‘DECI_KELVIN_TO_CELSIUS’ [-Werror=implicit-function-declaration]
 1698 |  value = DECI_KELVIN_TO_CELSIUS((value & 0xFFFF)) * 1000;
      |          ^~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:268: /home/utkarsh/GitHub/Others/faustus/src/faustus.o] Error 1
make[1]: *** [Makefile:1683: /home/utkarsh/GitHub/Others/faustus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.1-050601-generic'
make: *** [Makefile:9: default] Error 2

```

stwagner, 2020-04-04:
compile fix in https://github.com/hackbnw/faustus/pull/25

UtkarshVerma, 2020-04-05:
@stwagner Thanks, this worked for me but now `make dkms` returns an error:
```
$ sudo make dkms
Error! There are no instances of module: faustus
0.1 located in the DKMS tree.

------------------------------
Deleting module version: 0.2
completely from the DKMS tree.
------------------------------
Done.
dkms add .

Creating symlink /var/lib/dkms/faustus/0.2/source ->
                 /usr/src/faustus-0.2

DKMS: add completed.
dkms install -m faustus -v 0.2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j8 KERNELRELEASE=5.6.1-050601-generic -C /lib/modules/5.6.1-050601-generic/build SUBDIRS=/var/lib/dkms/faustus/0.2/build modules...(bad exit status: 2)
ERROR (dkms apport): binary package for faustus: 0.2 not found
Error! Bad return status for module build on kernel: 5.6.1-050601-generic (x86_64)
Consult /var/lib/dkms/faustus/0.2/build/make.log for more information.
make: *** [Makefile:24: dkms] Error 10
```
Also, this is the log:
```
DKMS make.log for faustus-0.2 for kernel 5.6.1-050601-generic (x86_64)
Sun Apr  5 08:39:59 IST 2020
make[1]: Entering directory '/usr/src/linux-headers-5.6.1-050601-generic'
  HOSTCC  scripts/basic/fixdep
  DESCEND  objtool
  UPD     include/generated/utsrelease.h
/usr/src/linux-headers-5.6.1-050601-generic/tools/build/Makefile.build:37: /usr/src/linux-headers-5.6.1-050601-generic/tools/build/Build.include: No such file or directory
make[5]: *** No rule to make target '/usr/src/linux-headers-5.6.1-050601-generic/tools/build/Build.include'.  Stop.
make[4]: *** [Makefile:43: /usr/src/linux-headers-5.6.1-050601-generic/tools/objtool/fixdep-in.o] Error 2
make[3]: *** [/usr/src/linux-headers-5.6.1-050601-generic/tools/build/Makefile.include:5: fixdep] Error 2
make[2]: *** [Makefile:68: objtool] Error 2
make[1]: *** [Makefile:1786: tools/objtool] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/usr/src/linux-headers-5.6.1-050601-generic'
```

stwagner, 2020-04-05:
Oh, yes that's an issue since kernel 5.4. I've no idea how to fix the dkms build.
As a quick workaround you can build and install the module with this commands:
#> make && make install
#> depmod
#> modprobe faustus

UtkarshVerma, 2020-04-06:
@stwagner I tried those commands, but I'm getting errors:
`make install`:
```
$ sudo make -C /lib/modules/$(uname -r)/build M=$(pwd) modules_install
make: Entering directory '/usr/src/linux-headers-5.6.2-050602-generic'
  INSTALL /home/utkarsh/GitHub/Others/faustus/src/faustus.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.6.2
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make: Leaving directory '/usr/src/linux-headers-5.6.2-050602-generic'
```
`depmod` runs fine, but `modprobe` gives error that it couldn't find the module faustus, and weirdly, `faustus.ko` is getting placed at `/lib/modules/5.6.2/build` rather than `/lib/modules/5.6.2-050602-generic/build`.
```
$ sudo modprobe faustus
modprobe: FATAL: Module faustus not found in directory /lib/modules/5.6.2-050602-generic

```

UtkarshVerma, 2020-04-27:
Closing because the patch mentioned in #26 worked for me.

closed 2020-04-27 by UtkarshVerma