hackbnw/faustus #26: DKMS build fails on Kubuntu 20.04 LTS AMD64 with 5.4.0-21-generic kernel

S-trace, 2020-04-17:
Hello.

Current master branch (d068a2b) fails to build in DKMS, but builds fine using make in the source directory directly:
```
Пт апр 17 12:23:11-s-trace@tuf:/usr/src/faustus-0.2
^_^ git describe --always
d068a2b

Пт апр 17 12:23:11-s-trace@tuf:/usr/src/faustus-0.2
^_^ LANG=C sudo dkms build faustus/0.2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.4.0-21-generic -C /lib/modules/5.4.0-21-generic/build SUBDIRS=/var/lib/dkms/faustus/0.2/build modules...(bad exit status: 2)
ERROR (dkms apport): binary package for faustus: 0.2 not found
Error! Bad return status for module build on kernel: 5.4.0-21-generic (x86_64)
Consult /var/lib/dkms/faustus/0.2/build/make.log for more information.

Пт апр 17 12:23:21-s-trace@tuf:/usr/src/faustus-0.2
O_O cat /var/lib/dkms/faustus/0.2/build/make.log
DKMS make.log for faustus-0.2 for kernel 5.4.0-21-generic (x86_64)
Fri Apr 17 12:23:19 MSK 2020
make: Entering directory '/usr/src/linux-headers-5.4.0-21-generic'
  DESCEND  objtool
/usr/src/linux-headers-5.4.0-21-generic/tools/build/Makefile.build:37: /usr/src/linux-headers-5.4.0-21-generic/tools/build/Build.include: No such file or directory
make[4]: *** No rule to make target '/usr/src/linux-headers-5.4.0-21-generic/tools/build/Build.include'.  Stop.
make[3]: *** [Makefile:43: /usr/src/linux-headers-5.4.0-21-generic/tools/objtool/fixdep-in.o] Error 2
make[2]: *** [/usr/src/linux-headers-5.4.0-21-generic/tools/build/Makefile.include:5: fixdep] Error 2
make[1]: *** [Makefile:67: objtool] Error 2
make: *** [Makefile:1823: tools/objtool] Error 2
make: *** Waiting for unfinished jobs....
make: Leaving directory '/usr/src/linux-headers-5.4.0-21-generic'

Пт апр 17 12:24:12-s-trace@tuf:/usr/src/faustus-0.2
^_^ LANG=C sudo make
make -C /lib/modules/5.4.0-21-generic/build M=/usr/src/faustus-0.2 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-21-generic'
  CC [M]  /usr/src/faustus-0.2/src/faustus.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /usr/src/faustus-0.2/src/faustus.mod.o
  LD [M]  /usr/src/faustus-0.2/src/faustus.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-21-generic'
```
Could you please help me fix this issue?

Thank you.

S-trace, 2020-04-17:
Fixed this issue with following patch:
```
diff --git a/dkms.conf b/dkms.conf
index 4b9ca4e..53c05ad 100644
--- a/dkms.conf
+++ b/dkms.conf
@@ -1,8 +1,8 @@
 PACKAGE_VERSION="0.2"
 
 PACKAGE_NAME="faustus"
-MAKE[0]="make -C ${kernel_source_dir} SUBDIRS=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build modules"
-CLEAN="make -C ${kernel_source_dir} SUBDIRS=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build clean"
+MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build modules"
+CLEAN="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build clean"
 
 BUILT_MODULE_NAME[0]="faustus"
 BUILT_MODULE_LOCATION[0]="src/"
```
Just replaced SUBDIRS to M

hackbnw, 2020-05-02:
@S-trace  The patch looks correct and seems to work on older kernels too, it is indeed the typical syntax according to https://wiki.debian.org/KernelDKMS Also I've noticed that the flags were inconsistently used, in `Makefile` it was `M=` already. I've commited the fix onto master. Thanks!

closed 2020-05-04 by hackbnw