hackbnw/faustus #27: Collaboration

flukejones, 2020-04-23:
Hi! I spotted this in one of your comments on the GOL thread for the [tool](https://github.com/flukejones/rog-core) I'm working on.

I'd love to shift some of the functionality I've got working into the kernel module and hopefully in to mainline kernel some day. I've done a fair bit of research, much of what I try to post to a bug tracked [here](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1802652). Maybe some of this could be useful to you?

hackbnw, 2020-05-02:
Hi @flukejones , I believe I'm not on (Gaming On Linux?). Thanks for reaching out! I've included your project in the README.

Some issues to consider about including support in kernel. Generally people have success with writing USB drivers in user space by looking on some other projects. This driver controls RGB via WMI. Not that it's not possible in user space (you can do direct calls via debugfs). The upside of kernel, one can potentially make unified ABI, downside, it is slower and more complex to implement.

Another issue to consider, there is currently active development of multicolor LED framework happening in Linux kernel, up until then, the ABI for RGB lighting is not really defined.

If you ask me, I'm not sure myself if the RGB part of this project should be a kernel module and it's a correct decision.

Also take a look at OpenRGB project, it seems to support quite a lot of devices at the time.

flukejones, 2020-05-05:
My main primary concern is not the RGB abilities at all, but more getting the Fn+ keys working correctly.

The GX502 and similar have:
- Volume + media keys + mic-mute
- Keyboard brightness,
- fan control / boost
- display keys
- touchpad enable/disable
- sleep,
- plane mode

Some also have extra Fn+Left/Right to change the builtin LED mode.

It looks like fan-boost is working for this model thanks to your work. But this model is slightly different in that the Volume + Media keys exist on a HID page for "Consumer Device", and the rest of the Fn+Keys exist on a second "Vendor Defined" HID page.

For example the volume+media keys are on page 0x02:
```
[0x02, 0xEA, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

while the fan is on page:

```
[0x5A, 0xAE, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
``

I quite honestly haven't the first clue to using C, much less working with kernel. I imagine I can provide you with almost all information required for this laptop series.