hackbnw/faustus #28: Brightness controls no longer work

Xabifk, 2020-04-26:
`xbacklight` changes the the value in `/sys/class/backlight/faustus/brightness` but my screen is only affected by changes in `/sys/class/backlight/amdgpu_bl0/brightness`   
If I remove faustus, xbacklight works as intended.  
  
I use an Asus FX505DT

```
$ uname -a
Linux Maximegalon 5.6.6-arch1-1 #1 SMP PREEMPT Tue, 21 Apr 2020 10:35:16 +0000 x86_64 GNU/Linux
$ modinfo faustus
filename:       /lib/modules/5.6.6-arch1-1/extra/faustus.ko.xz
license:        GPL
description:    Backport of Asus Generic WMI Driver
author:         Corentin Chary <corentin.chary@gmail.com>, Yong Wang <yong.y.wang@intel.com>
srcversion:     BD28779796F3429299C4D1F
depends:        wmi,sparse-keymap,rfkill
retpoline:      Y
name:           faustus
vermagic:       5.6.6-arch1-1 SMP preempt mod_unload
parm:           let_it_burn:Disable DMI check, force load (bool)
```
I also made the changes that make compilation possible on my kernel version.

hackbnw, 2020-05-02:
Could you try updating? I've merged linux mainline. The backlight should behave exactly the same as in default platform driver asus_wmi.

Xabifk, 2020-05-02:
It doesn't seem to work. I don't know if it's a configuration problem on my part.\n
xbacklight uses faustus, but the rest of the system does not

hackbnw, 2020-05-04:
Ok, don't know what the issue is with xbacklight, fun thing being I don't have asus backlight device at all. It looks like you could try providing correct configuration in Xorg config: <https://wiki.archlinux.org/index.php/Backlight#xbacklight>

Also I have found that it always reverts to backlight type "vendor" if you don't have CONFIG_ACPI_VIDEO enabled, I don't known if you configured the kernel yourself or is some standard OS kernel?
 
```
$ cat /boot/config-<version>| grep CONFIG_ACPI_VIDEO
CONFIG_ACPI_VIDEO=m
```

Either way this should also be exactly the same when you use default driver.

Xabifk, 2020-05-06:
I don't have a config file there. I really don't know what happened, it just stopped working after a kernel update - I think

As I'm the only one affected by this you can close the issue if you want. I'll come back with a solution if I found one. For now I just manually write to `/sys/class/backlight/amdgpu_bl0/brightness`

RadoiAndrei, 2020-06-28:
Same issue for me on my FX505DT.