hackbnw/faustus #30: debian buster  make dkms drops an arror

niehausCNC, 2020-05-01:
Hallo,
get the fancontrol and backlight control running on buster but the permanent installation fails. 
Insallationprocess described at:
[debianuser](http://forums.debian.net/viewtopic.php?f=5&t=146104)

Here the some information I get from my system:
` sudo make dkms
`
```
Error! There are no instances of module: faustus
0.1 located in the DKMS tree.

------------------------------
Deleting module version: 0.2
completely from the DKMS tree.
------------------------------
Done.
dkms add .

Creating symlink /var/lib/dkms/faustus/0.2/source ->
                 /usr/src/faustus-0.2

```
`/var/lib/dkms/faustus/0.2/build/make.log `
```
DKMS make.log for faustus-0.2 for kernel 5.4.0-0.bpo.4-amd64 (x86_64)
Thu Apr 30 11:47:46 CEST 2020
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-0.bpo.4-amd64'
sh: 0: Can't open /usr/src/linux-headers-5.4.0-0.bpo.4-common/scripts/mkmakefile
make[3]: *** [/usr/src/linux-headers-5.4.0-0.bpo.4-common/Makefile:513: outputmakefile] Error 127
make[3]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-5.4.0-0.bpo.4-common/scripts/Makefile.build:42: /usr/src/linux-headers-5.4.0-0.$
make[4]: *** No rule to make target '/usr/src/linux-headers-5.4.0-0.bpo.4-common/scripts/basic/Makefil$
make[3]: *** [/usr/src/linux-headers-5.4.0-0.bpo.4-common/Makefile:499: scripts_basic] Error 2
make[2]: *** [/usr/src/linux-headers-5.4.0-0.bpo.4-common/Makefile:677: include/config/auto.conf.cmd] $
make[1]: *** [/usr/src/linux-headers-5.4.0-0.bpo.4-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-0.bpo.4-amd64'

```
```
sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3 BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX705DY.314
        Release Date: 11/01/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX705DY
        Version: 1.0       

```
```
System:    Host: reddeck Kernel: 5.4.0-0.bpo.4-amd64 x86_64 bits: 64 compiler: gcc v: 8.3.0
           Desktop: KDE Plasma 5.14.5 tk: Qt 5.11.3 wm: kwin_x11 dm: SDDM
           Distro: Debian GNU/Linux 10 (buster)
Machine:   Type: Laptop System: ASUSTeK product: TUF Gaming FX705DY_FX705DY v: 1.0 serial: <filter>
           Mobo: ASUSTeK model: FX705DY v: 1.0 serial: <filter> UEFI: American Megatrends
           v: FX705DY.314 date: 11/01/2019 
```
  

```
    [ 3618.190836] [drm] VCE initialized successfully.
    [ 3624.893041] amdgpu 0000:01:00.0: GPU pci config reset
    [ 5572.146656] faustus: DMI checK: FX705DY
    [ 5572.146888] faustus: Initialization: 0x1
    [ 5572.146991] faustus: BIOS WMI version: 9.0
    [ 5572.147053] faustus: SFUN value: 0x21
    [ 5572.147057] faustus faustus: Use DSTS
    [ 5572.147059] faustus faustus: Enable event queue
    [ 5572.149451] input: Asus WMI hotkeys as /devices/platform/faustus/input/input17
    [ 5572.149722] faustus: Number of fans: 0
```
so fn + arror up lights up the keyboard
and 
fn+F5
adjust  the fanmode

But I do not unterstand what is with` make dkms` wrong.
Thank you for your work.

hackbnw, 2020-05-02:
Hello, please try updating, the issue should now be fixed, it's likely the same issue as #26 and #12

niehausCNC, 2020-05-04:
Yes it worked!
`sudo make onboot`
worked on debian buster too.
Thank you very much!

```
 sudo make dkms
Error! There are no instances of module: faustus
0.1 located in the DKMS tree.

------------------------------
Deleting module version: 0.2
completely from the DKMS tree.
------------------------------
Done.
dkms add .

Creating symlink /var/lib/dkms/faustus/0.2/source ->
                 /usr/src/faustus-0.2

DKMS: add completed.
dkms install -m faustus -v 0.2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.4.0-0.bpo.4-amd64 -C /lib/modules/5.4.0-0.bpo.4-amd64/build M=/var/lib/dkms/faustus/0.2/build modules...
cleaning build area...

DKMS: build completed.

faustus.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-0.bpo.4-amd64/updates/dkms/

depmod...

Backing up initrd.img-5.4.0-0.bpo.4-amd64 to /boot/initrd.img-5.4.0-0.bpo.4-amd64.old-dkms
Making new initrd.img-5.4.0-0.bpo.4-amd64
(If next boot fails, revert to initrd.img-5.4.0-0.bpo.4-amd64.old-dkms image)
update-initramfs.......

DKMS: install completed.
root@reddeck:/home/hn/linuxstuff/faustus-master# 
```

closed 2020-05-04 by hackbnw