hackbnw/faustus #31: Fan mode file changed

Xabifk, 2020-05-02:
The fan mode file has changed from `/sys/devices/platform/faustus/fan_mode` to `/sys/devices/platform/faustus/fan_boost_mode`

yaliv, 2020-05-03:
> The fan mode file has changed from `/sys/devices/platform/faustus/fan_mode` to `/sys/devices/platform/faustus/fan_boost_mode`

Confirm that.
But although I hear speed changes to my fan after pressing `Fn`+`F5`, the value of `/sys/devices/platform/faustus/fan_boost_mode` is always `0`.
:man_shrugging:

hackbnw, 2020-05-04:
The new name is correct one, this is how it's named in the mainline driver and by the vendor, as opposed to the fan_mode which usually means automatic or manual. It was an error on my part. Sorry for that. I will update the README.

hackbnw, 2020-05-04:
@yaliv are you sure? what does it say in `dmesg`, should be the following

```
[96234.887932] faustus: Set fan boost mode: 0
[96236.838525] faustus: Set fan boost mode: 1
[96237.708032] faustus: Set fan boost mode: 2
[96238.682679] faustus: Set fan boost mode: 0
```

hackbnw, 2020-05-04:
@yaliv The mainline driver has separate option for your laptop which is named `throttle_thermal_policy` which has only silent and default mode if I'm correct. You should have `/sys/devices/platform/faustus/throttle_thermal_policy` file where the value changes (probably there might be laptops with both options present simultaneously?)

Xabifk, 2020-05-04:
@hackbnw I can confirm that's how it works for me
Maybe add a note in the README?

hackbnw, 2020-05-04:
Thanks for noticing! Just updated the README.

yaliv, 2020-05-05:
> @yaliv The mainline driver has separate option for your laptop which is named `throttle_thermal_policy` which has ~only silent and default mode~ always 3 modes. You should have `/sys/devices/platform/faustus/throttle_thermal_policy` file where the value changes (probably there might be laptops with both options present simultaneously?)

Great! I can see value changes in `throttle_thermal_policy`:
```sh
$ cat /sys/devices/platform/faustus/throttle_thermal_policy
0
$ cat /sys/devices/platform/faustus/throttle_thermal_policy
1
$ cat /sys/devices/platform/faustus/throttle_thermal_policy
2
```
:+1:

Xabifk, 2020-05-06:
As the README was updated and everything seems clear I'll close this issue

closed 2020-05-06 by Xabifk