hackbnw/faustus #32: FR: Make `showcodes` or `xev` register `Fn` + `F5` keypress.

UtkarshVerma, 2020-05-03:
The driver is working really well for me, however, there's one small thing that I really miss. Currently, `Fn`+`F5` doesn't trigger any scancode or keycode. I want to bind this key combination using `i3` to display OSD on Kubuntu, therefore please consider implementing this as I think it would be very handy to conveniently see the CPU mode on screen rather than running `dmesg` every time.

> This is the way I'm planning to show OSD:
  ```qdbus org.kde.kscreen.osdService /org/kde/kscreen/osdService showOsd "sensors-fan-symbolic" "Performance"```

hackbnw, 2020-05-04:
Here you go, update and add following to 

```text
$ cat /etc/modprobe.d/faustus.conf 
blacklist asus_wmi
blacklist asus_nb_wmi
options faustus report_key_events=1
```

Unfortunately I'm not seeing it making it to the mainline, as this event code conflicts with the "phone" key on some old laptop. I had this option included initially. Also is there like a special scan key for fan mode? I did bound it to FN_F5 for now.

hackbnw, 2020-05-04:
Also can recommend `evtest` for testing, it is rather complex how X handles the input devices.

UtkarshVerma, 2020-05-05:
> Here you go, update and add following to
> 
> ```
> $ cat /etc/modprobe.d/faustus.conf 
> blacklist asus_wmi
> blacklist asus_nb_wmi
> options faustus report_key_events=1
> ```
> 
> Unfortunately I'm not seeing it making it to the mainline, as this event code conflicts with the "phone" key on some old laptop. I had this option included initially. Also is there like a special scan key for fan mode? I did bound it to FN_F5 for now.

`FN_F5` is triggering keycode `470`(found using `showkey` in virtual console). Now my issue is solved since to make X respect this keypress, all I need to do is assign a keycode between `8-255`. 
I'm planning to write an article on this, but for now, this is how I'm planning to do so:
https://unix.stackexchange.com/questions/575515/xev-doesnt-register-my-fnf4-keypress-how-to-bind-this-combination

UtkarshVerma, 2020-05-05:
@hackbnw On my machine, ASUS FX505DT, the scan code for this key is `0xae`.

UtkarshVerma, 2020-09-10:
Adding `options faustus report_key_events=1` to `faustus.conf` was sufficient to make `evtest` detect the keypress. After that, I used `udev` to set the keycode for this key to 248 so that it falls within X's input range, 0-255.

closed 2020-09-10 by UtkarshVerma