hackbnw/faustus #33: Not working on Asus FX505DT

TaherPatan, 2020-05-05:
I am using POP OS 20.04 which is based on Ubuntu 20.04 and it s not working, I am not able to control the fan speed using it so please help me control it.
The information is attached below, if you need anything additional I can provide:
```
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FX505DT.310
	Release Date: 12/24/2019
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FX505DT
	Version: 1.0

```

Syzygianinfern0, 2020-05-08:
```bash
cat /sys/devices/platform/faustus/throttle_thermal_policy
```

What is your output?

TaherPatan, 2020-05-10:
> ```shell
> cat /sys/devices/platform/faustus/throttle_thermal_policy
> ```
> 
> What is your output?

It shows 0

JORGETECH, 2020-05-15:
Weird, I just installed Faustus as dkms on Ubuntu 20.04 and it works perfectly.

I can set the fan speed with "Fn-F5" and I can also set the color using the "set_rgb.sh" script and running OpenRGB as root.

I am using the same laptop model (FX-505DT) with BIOS version 310. Mine has the 60 Hz panel though.

JORGETECH, 2020-05-15:
`cat /sys/devices/platform/faustus/throttle_thermal_policy`

Shows number 2 for me

JORGETECH, 2020-05-20:
I forgot to put the BIOS info so here it is:

```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505DT.310
        Release Date: 12/24/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505DT
        Version: 1.0
```

cromerc, 2020-09-10:
@UtkarshVerma Try this fork: https://github.com/rublag/faustus

It has fan mode polling, so you can poll /sys/devices/platform/faustus/throttle_thermal_policy to be notified when the fan mode changes.

Or if you just want an easy way to change colors, fan mode, and have notifications, check out TUF Manager: https://git.cromer.cl/cromer/tuf-manager

TUF Manager let's you control the keyboard colors, fan mode, restores config on login, and comes with a status bar icon that pops up a notification when the fan mode changes. TUF Manager also requires the fork at https://github.com/rublag/faustus because it uses the fan mode polling to see if the mode changed.

UtkarshVerma, 2020-09-10:
@cromerc Thanks for referring the software to me. It looks great. For now, I've just made things work out using this repo itself. Using #32 I enabled the key reporting in faustus, then using `udev` I added  a rule to map the `Fn-F5` key's scan code, `0xae` for me, to keycode 240.
Since I use i3wm on my Kubuntu install, binding this keypress was trivial and I wrote a BASH script to read `/sys/devices/platform/faustus/throttle_thermal_policy` and show an OSD appropriately.

In Kubuntu, I could easily show the OSD using:
```
 qdbus org.kde.kscreen.osdService /org/kde/kscreen/osdService showOsd "sensors-fan-symbolic" "Hi"
```

This is the flow I've settled on right now and it works pretty nicely.

ferretwithaberet, 2021-01-19:
[This](https://gist.github.com/ferretwithaberet/4e3e1bb7a8db57e8282696ae34d94c5d/) is a quick utility I made helped by @rublag and @cromerc.
The polling part(throttle-policy-poller.service and throttle_policy_poller.py) only works on [Rublag's fork](https://github.com/rublag/faustus) which has polling implemented.
You can run `get_throttle_policy.py` by using a shortcut like `Ctrl+Shift+F5`(This is what I am using) to run that python file which will send you a notification with the current running throttle policy mode.

closed 2021-03-14 by hackbnw