hackbnw/faustus #34: Can't control rgb on fx505dt

n1gh7maan, 2020-05-10:
I can't control for some reason my keyboard. Every file in  `/sys/devices/platform/faustus/kbbl` has 00 value. When i try to change any file in that directory, the value switches back to 00. After installation of the driver I was able to properly control leds. 
OS: Manjaro 20.04
kernel: 5.6.8-1-MANJARO
FX505DT

n1gh7maan, 2020-05-11:
Nevermind, I figured it out. I forgot to set kbbl_flags. 
Is it just me or rgb lights in bios are brighter than in windows/linux?

TaherPatan, 2020-05-11:
> Nevermind, I figured it out. I forgot to set kbbl_flags.
> Is it just me or rgb lights in bios are brighter than in windows/linux?

Are you able to control the fans?

n1gh7maan, 2020-05-11:
> Are you able to control the fans?
yes

TaherPatan, 2020-05-11:
> > Are you able to control the fans?
> 
> yes

You have asus fx505dt with gtx 1650 right? Did you change any settings to be able to control fans? Is there any indication that the fan speed changed?

n1gh7maan, 2020-05-11:
for me fn+f5 works after installing this driver. You can check whether pressing these buttons work by running `cat /sys/devices/platform/faustus/throttle_thermal_policy`

closed 2020-05-15 by n1gh7maan