hackbnw/faustus #35: Support for FA* series, specifically FA506II

lknop, 2020-05-13:
I have just installed linux on a new TUF laptop and I think the faustus module might be possible to be extended for the FA* series. 

When trying to insmod, I get the following:
```
[ 1454.866709] faustus: Omitting DMI verification
[ 1454.866803] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
[ 1454.866807] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
[ 1454.866825] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
[ 1454.866827] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
[ 1454.866844] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
[ 1454.866846] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
[ 1454.866850] faustus faustus: Use DSTS
[ 1454.866851] faustus faustus: Enable event queue
[ 1454.866883] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
[ 1454.866885] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
[ 1454.866902] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
[ 1454.866903] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
[ 1454.866908] faustus: probe of faustus failed with error -5
[ 1454.866917] faustus: Can't probe platform driver: -19
```

The DMI bios/board versions are:
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA506II.304
        Release Date: 03/24/2020
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA506II
        Version: 1.0
```

I also attach the dsdt.aml
[dsdt.aml.zip](https://github.com/hackbnw/faustus/files/4622776/dsdt.aml.zip)

Can you give more pointers how to help in identifying the issue and possibly adding support for my hardware? I can also try changing the code, but having no experience in linux kernel I am not sure where to start.

xurei, 2020-05-22:
I have a FA506II.302 myself. Did you manage to make it work ?

lknop, 2020-05-22:
> I have a FA506II.302 myself. Did you manage to make it work ?

Unfortunately no, I didn't. I got to the point of installing windows just to disable the blinking RGB, and spent over a day trying to make it work under windows. So far I only have red backlight, but the asus software fails to start. I tried OpenRGB under windows to sniff the i2c bus but no luck.

xurei, 2020-05-22:
I can't PM him either, but I believe I can try a few things given his messages.

To be continued... ;-)

xurei, 2020-05-24:
I made some progress on stabilizing the beast... I have the Nvidia driver working through prime-run.

@lknop did you try to edit the code to match the devide descriptions ? 

I am not sure it's safe to run it... @hackbnw, any remark ?
If you think it's safe I can try it and give feedback.

``` 
(line 3250 in branch 08f44119b93710d9c3137f6cd52e064ea7b15b03)
	{
		.callback = dmi_check_callback,
		.ident = "FA506II",
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "ASUSTeK COMPUTER INC."),
			DMI_MATCH(DMI_PRODUCT_NAME, "FA506II"),
		},
	},
```

lknop, 2020-05-25:
@xurei no, I didn't edit the code but I used the flag that skips the DMI verification 

`sudo insmod ./src/faustus.ko let_it_burn=1` 

The log I pasted in the issue report is from that command as faustus nicely logged

`[ 1454.866709] faustus: Omitting DMI verification`

So I cannot say it's safe but it didn't break anything in my setup, just didn't work.

rublag, 2020-06-03:
I managed to make it work. My laptop is TUF Gaming FX706II.

dmidecode info:
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA706II.304
...
System Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF Gaming FA706II_FX706II
...
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA706II
```

Module asus-nb-wmi (or faustus) cannot load and prints error:
```
ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
```

It occurs because of the difference between acpi code.
ASL code for working models:
```
CreateDWordField (Arg2, Zero, IIA0)
CreateDWordField (Arg2, 0x04, IIA1)
CreateDWordField (Arg2, 0x08, IIA2)
```

ASL code for my model:
```
CreateDWordField (Arg2, Zero, IIA0)
CreateDWordField (Arg2, 0x04, IIA1)
CreateDWordField (Arg2, 0x08, IIA2)
CreateDWordField (Arg2, 0x0C, IIA3)
CreateDWordField (Arg2, 0x10, IIA4)
```

It is similar to this issue https://patchwork.kernel.org/patch/10909549/

So I edited faustus.c like in the patch above

```diff
diff --git a/src/faustus.c b/src/faustus.c
index 39c1f19..814d242 100644
--- a/src/faustus.c
+++ b/src/faustus.c
@@ -150,6 +150,8 @@ struct bios_args {
        u32 arg0;
        u32 arg1;
        u32 arg2; /* At least TUF Gaming series uses 3 dword input buffer. */
+       u32 arg3;
+       u32 arg4;
 } __packed;
 
 /*
@@ -315,13 +317,15 @@ static void asus_wmi_input_exit(struct asus_wmi *asus)
 
 /* WMI ************************************************************************/
 
-static int asus_wmi_evaluate_method3(u32 method_id,
-               u32 arg0, u32 arg1, u32 arg2, u32 *retval)
+static int asus_wmi_evaluate_method5(u32 method_id,
+               u32 arg0, u32 arg1, u32 arg2, u32 arg3, u32 arg4, u32 *retval)
 {
        struct bios_args args = {
                .arg0 = arg0,
                .arg1 = arg1,
                .arg2 = arg2,
+               .arg3 = arg3,
+               .arg4 = arg4,
        };
        struct acpi_buffer input = { (acpi_size) sizeof(args), &args };
        struct acpi_buffer output = { ACPI_ALLOCATE_BUFFER, NULL };
@@ -350,6 +354,12 @@ static int asus_wmi_evaluate_method3(u32 method_id,
        return 0;
 }
 
+static int asus_wmi_evaluate_method3(u32 method_id, 
+               u32 arg0, u32 arg1, u32 arg2, u32 *retval)
+{
+       return asus_wmi_evaluate_method5(method_id, arg0, arg1, arg2, 0, 0, retval);
+}
+
 static int asus_wmi_evaluate_method(u32 method_id, u32 arg0, u32 arg1, u32 *retval)
 {
        return asus_wmi_evaluate_method3(method_id, arg0, arg1, 0, retval);
```

Now keyboard backlight management works.

But there are problems during loading:
```
faustus: Failed to set throttle thermal policy (retval): 0x0
battery: extension failed to load: ASUS Battery Extension
battery: extension unregistered: ASUS Battery Extension
``` 

When calling `set_rgb.sh` following error occurs (but color changes):
```
faustus: RGB keyboard device 1, write error (retval): 0
```

When pressing Aura next/previous key (Fn-Left/Fn-Right) it says about unknown keys:
```
faustus: Unknown key b3 pressed
```

P. S. If display brightness control doesn't work, you can use this patch (https://patchwork.freedesktop.org/patch/367486/)
```diff
diff --git a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
index ddc979e3eebe..acd4874e0743 100644
--- a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
+++ b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
@@ -1377,13 +1377,10 @@ static int dm_late_init(void *handle)
         */
        params.min_abm_backlight = 0x28F;
 
-       /* todo will enable for navi10 */
-       if (adev->asic_type <= CHIP_RAVEN) {
-               ret = dmcu_load_iram(dmcu, params);
+       ret = dmcu_load_iram(dmcu, params);
 
-               if (!ret)
-                       return -EINVAL;
-       }
+       if (!ret)
+               return -EINVAL;
 
        return detect_mst_link_for_all_connectors(adev->ddev);
 }
```

rublag, 2020-06-03:
I managed to make it work. My laptop is TUF Gaming FX706II.

dmidecode info:
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA706II.304
...
System Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF Gaming FA706II_FX706II
...
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA706II
```

Module asus-nb-wmi (or faustus) cannot load and prints error:
```
ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200110/dsopcode-198)
ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200110/psparse-529)
```

It occurs because of the difference between acpi code.
ASL code for working models:
```
CreateDWordField (Arg2, Zero, IIA0)
CreateDWordField (Arg2, 0x04, IIA1)
CreateDWordField (Arg2, 0x08, IIA2)
```
ASL code for my model:
```
CreateDWordField (Arg2, Zero, IIA0)
CreateDWordField (Arg2, 0x04, IIA1)
CreateDWordField (Arg2, 0x08, IIA2)
CreateDWordField (Arg2, 0x0C, IIA3)
CreateDWordField (Arg2, 0x10, IIA4)
```

It is similar to this issue https://patchwork.kernel.org/patch/10909549/

So I edited faustus.c like in the patch above
```diff
diff --git a/src/faustus.c b/src/faustus.c
index 39c1f19..814d242 100644
--- a/src/faustus.c
+++ b/src/faustus.c
@@ -150,6 +150,8 @@ struct bios_args {
        u32 arg0;
        u32 arg1;
        u32 arg2; /* At least TUF Gaming series uses 3 dword input buffer. */
+       u32 arg3;
+       u32 arg4;
 } __packed;
 
 /*
@@ -315,13 +317,15 @@ static void asus_wmi_input_exit(struct asus_wmi *asus)
 
 /* WMI ************************************************************************/
 
-static int asus_wmi_evaluate_method3(u32 method_id,
-               u32 arg0, u32 arg1, u32 arg2, u32 *retval)
+static int asus_wmi_evaluate_method5(u32 method_id,
+               u32 arg0, u32 arg1, u32 arg2, u32 arg3, u32 arg4, u32 *retval)
 {
        struct bios_args args = {
                .arg0 = arg0,
                .arg1 = arg1,
                .arg2 = arg2,
+               .arg3 = arg3,
+               .arg4 = arg4,
        };
        struct acpi_buffer input = { (acpi_size) sizeof(args), &args };
        struct acpi_buffer output = { ACPI_ALLOCATE_BUFFER, NULL };
@@ -350,6 +354,12 @@ static int asus_wmi_evaluate_method3(u32 method_id,
        return 0;
 }
 
+static int asus_wmi_evaluate_method3(u32 method_id, 
+               u32 arg0, u32 arg1, u32 arg2, u32 *retval)
+{
+       return asus_wmi_evaluate_method5(method_id, arg0, arg1, arg2, 0, 0, retval);
+}
+
 static int asus_wmi_evaluate_method(u32 method_id, u32 arg0, u32 arg1, u32 *retval)
 {
        return asus_wmi_evaluate_method3(method_id, arg0, arg1, 0, retval);
```

Now keyboard backlight management works.

But there are problems during loading:
```
faustus: Failed to set throttle thermal policy (retval): 0x0
battery: extension failed to load: ASUS Battery Extension
battery: extension unregistered: ASUS Battery Extension
```

When calling set_rgb.sh following error occurs (but color changes):
```
faustus: RGB keyboard device 1, write error (retval): 0
```

When pressing Aura next/previous key (Fn-Left/Fn-Right) it says about unknown keys:
```
faustus: Unknown key b3 pressed
```

P. S. If display brightness control doesn't work, you can use this patch (https://patchwork.freedesktop.org/patch/367486/)
```diff
diff --git a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
index ddc979e3eebe..acd4874e0743 100644
--- a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
+++ b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
@@ -1377,13 +1377,10 @@ static int dm_late_init(void *handle)
         */
        params.min_abm_backlight = 0x28F;
 
-       /* todo will enable for navi10 */
-       if (adev->asic_type <= CHIP_RAVEN) {
-               ret = dmcu_load_iram(dmcu, params);
+       ret = dmcu_load_iram(dmcu, params);
 
-               if (!ret)
-                       return -EINVAL;
-       }
+       if (!ret)
+               return -EINVAL;
 
        return detect_mst_link_for_all_connectors(adev->ddev);
 }
```

xurei, 2020-06-05:
@rublag Do you have a fork so I can try it on my machine ?

rublag, 2020-06-05:
@xurei Not yet. You can just apply patch yourself. Or I can make a fork with the patch in a few days.

Hyper-KVM, 2020-06-06:
I wish I knew how to debug this so I could try changing the address values to the correct ones in faustus.c. the address `0x00120075` does not exist on FA506IV, so throttle_thermal_policy returns 0x0 and fails. I have no idea which address controls the throttle_thermal_policy lol but having that working would be nice to have hopefully better clock speeds on battery than default.

rublag, 2020-06-06:
@Hyper-KVM In faustus there are two methods for fan control: throttle thermal policy and fan boost mode. 

Maybe just try to swap values for `NOTIFY_KBD_TTP` and `NOTIFY_KBD_FBM` in faustus.c?

rublag, 2020-06-06:
@Hyper-KVM On my laptop `# echo 2 > /sys/devices/platform/faustus/fan_boost_mode` works without error. So it seems that ASUS changed keycode and we need to patch `NOTIFY_KBD_FBM` to use keycode `0xae` which is now used by `NOTIFY_KBD_TTP`.

Also I don't know why, but check for thermal throttle policy passes. Probably bug in faustus driver.

rublag, 2020-06-06:
@xurei I made a fork: https://github.com/rublag/faustus

Hyper-KVM, 2020-06-06:
> Also I don't know why, but check for thermal throttle policy passes. Probably bug in faustus driver.

In dmesg I get: `faustus: Failed to set throttle thermal policy (retval): 0x0`

If i change the value of NOTIFY_KBD_FBM to 0xae, It does say in dmesg it applied the fan boost mode but it doesn't seem to work in practice when stress testing? 

> I made a fork: https://github.com/rublag/faustus

Also, the fork does work, i get this in dmesg:
`faustus: Fan boost mode check successful
faustus: Fan boost mode check end
faustus: TTP check successful
faustus: TTP check end`
the TTP check is weird because address 0x00120075 does not exist in my DSDT.

rublag, 2020-06-06:
@Hyper-KVM Yes I've added these debug mesasges to understand what's going on. I don't know how fan boost should work. But yes, I don't see any effect and in ACPI table this function seems to have no effect:
```
Method (SFBM, 1, NotSerialized)
{
    If ((Arg0 == Zero)){}
    If ((Arg0 == One)){}
    If ((Arg0 == 0x02)){}
}
```
On one of the supported laptops this function is ~ 100 lines long. 

Does fan boost work on Windows? How can we test if it works properly?

Hyper-KVM, 2020-06-06:
> Does fan boost work on Windows? How can we test if it works properly?

@rublag As far as i can tell, all 3 fan modes work correctly. Using the armory crate software of course, not sure if that has anything to do with it.

xurei, 2020-06-10:
@rublag @Hyper-KVM Fan boost does work on Windows for the FA506II as well.

Hyper-KVM, 2020-06-11:
OK So a temporary solution @rublag @xurei, I changed the expected return value of setting throttle_thermal_policy from 1 to 0, and it works! the fan modes work! 
`From b742c7da481989b81c4dbf0568d8fdb040e5407a Mon Sep 17 00:00:00 2001
From: Hyper-KVM <hyperkvmx86@gmail.com>
Date: Thu, 11 Jun 2020 18:59:35 -0400
Subject: [PATCH] Change expected retval from 1 to 0

---
 src/faustus.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/faustus.c b/src/faustus.c
index 01e1fcf..f9261cc 100644
--- a/src/faustus.c
+++ b/src/faustus.c
@@ -2143,7 +2143,7 @@ static int throttle_thermal_policy_write(struct asus_wmi *asus)
                return err;
        }

-       if (retval != 1) {
+       if (retval != 0) {
                pr_warn("Failed to set throttle thermal policy (retval): 0x%x\n",
                        retval);
                return -EIO;
--
2.27.0
`

lknop, 2020-06-12:
@xurei @rublag I have created a fork with the above patch from @Hyper-KVM.

@Hyper-KVM Have you figured out which thermal policy corresponds to which fan mode? 
I also have broken display brightness, is the amdgpu_dm patch you posted available anywhere or do I need to compile the kernel myself? I am running manjaro at the moment.

rublag, 2020-06-12:
@lknop @Hyper-KVM 
Strange, but it really seems to work. Mode 2 is more quiet than others.
I think I'll research this more in July.

You need to recompile kernel
https://wiki.archlinux.org/index.php/Kernel/Arch_Build_System
Here is the guide.

Save patch as backlight.patch

and make these two changes in PKGBUILD:
```
source=(
  "$_srcname::git+https://git.archlinux.org/linux.git?signed#tag=$_srctag"
  config         # the main kernel config file
  sphinx-workaround.patch
  backlight.patch
)
```
and
```
sha256sums=('SKIP'
            '2a157fdbf3a6396e985db9ae5d11870a786717dca31de78cad09c06eb28761ff'
            '8cb21e0b3411327b627a9dd15b8eb773295a0d2782b1a41b2a8839d1b2f5778c'
            'SKIP')
```

Hyper-KVM, 2020-06-12:
>@Hyper-KVM Have you figured out which thermal policy corresponds to which fan mode?

@lknop The fan modes correspond to those in the readme on my machine.

Hyper-KVM, 2020-06-12:
>Strange, but it really seems to work. Mode 2 is more quiet than others.
I think I'll research this more in July.

I think it's due to the fact address `0x00120075` doesn't exist in the DSDT, so it will always return 0, but the address is present? It is strange indeed, There must be some other address that controls it now but I'm not experienced enough to figure out which, that patch I made was a shot in the dark. But I do look forward to what you come up with!

Hyper-KVM, 2020-06-13:
Made a fork with support for all FA506* and also now takes 1 or 0 as a value instead of 0.

xurei, 2020-06-15:
@Hyper-KVM I just tested your fork with my FA506II.

dmesg after modprobe:
```
[  273.084500] faustus: DMI checK: FA506II
[  273.084639] faustus: Initialization: 0x1
[  273.084693] faustus: BIOS WMI version: 8.1
[  273.084738] faustus: SFUN value: 0x4a0061
[  273.084740] faustus faustus: Use DSTS
[  273.084740] faustus faustus: Enable event queue
[  273.085220] faustus: Fan boost mode check successful
[  273.085220] faustus: Fan boost mode check end
[  273.086022] faustus: TTP check successful
[  273.086024] faustus: TTP check end
[  273.086764] faustus: Set throttle thermal policy mode: 0
[  273.089486] input: Asus WMI hotkeys as /devices/platform/faustus/input/input24
[  273.098413] battery: extension failed to load: ASUS Battery Extension
[  273.098416] battery: extension unregistered: ASUS Battery Extension
```

- Keyboard Backlight is working. 
- Battery extension fails to load.
- Fn-F5 gets captured by the driver and changes the thermal policy, but it doesn't seem to do anything. Fans are still in silent mode.

```
[  733.242247] faustus: Set throttle thermal policy mode: 0
[  733.242247] faustus: Set throttle thermal policy mode: 1
[  733.242247] faustus: Set throttle thermal policy mode: 2
```

Here is my dstd file for completeness.
[dsdt.zip](https://github.com/hackbnw/faustus/files/4780466/dsdt.zip)

Hyper-KVM, 2020-06-23:
There is a BIOS update available that has broken cpu_fan RPM reading (not that it matters since it returned 0 anyways, it just now does not show up) but at least now most IIA0 calls will return 0xFFFFFFFE aka UNSUPPORTED, this should make it easier to narrow down the correct addresses for everything.

 So far there is CRFC (0x00120072) that is not defined in faustus.h, when I tried to see if that address would do anything with ASUS_WMI_DEVID_THROTTLE_THERMAL_POLICY it returned a value of -19, so I'm not sure what that means hopefully it means something to someone lol. anyway I will attach both the diff of BIOS 308 vs 304 and my full dsdt for 308. 
[dsdtv308FA506IV.tar.gz](https://github.com/hackbnw/faustus/files/4816539/dsdtv308FA506IV.tar.gz)

rublag, 2020-06-29:
Some news. 
1. It seems that armoury crate doesn't use relevant WMI ACPI calls (but I am not very sure yet)
2. I managed to make cpu fan speed reading work. Same for gpu fan will not be hard.

rublag, 2020-06-30:
Fixed cpu fan reporting and battery error on my fork
https://github.com/rublag/faustus

Hyper-KVM, 2020-06-30:
> Fixed cpu fan reporting and battery error on my fork
> https://github.com/rublag/faustus

So I just tested this and battery module is loaded, but cpu_fan is output as N/A, but at least it is there I guess. but definetly making progress, I may try just changing the CPU FAN value from 0x00110013 to 0x00110014 to see if that works.

rublag, 2020-06-30:
It is strange because in your dsdt there is `0x110013` and it looks like mine. So it should work. Did you try to read `/sys/devices/platform/faustus/hwmon/hwmon5/fan1_input`?

Hyper-KVM, 2020-06-30:
> It is strange because in your dsdt there is `0x110013` and it looks like mine. So it should work. Did you try to read `/sys/devices/platform/faustus/hwmon/hwmon5/fan1_input`?

`cat /sys/devices/platform/faustus/hwmon/hwmon5/fan1_input`
returns nothing. Although, changing `asus->fan_type = FAN_TYPE_SPEC83;` to `asus->fan_type = FAN_TYPE_AGFN;` does give a value of 0 instead so I'm not sure what the issue is.

Hyper-KVM, 2020-07-22:
I added gpu fan sensor to my fork
https://github.com/Hyper-KVM/faustus