hackbnw/faustus #37: Works on FX705DT

ATechnoHazard, 2020-05-19:
I've tested the latest version on `FX705DT`, bios version `FX705DT.308` running `Arch Linux` with kernel `5.6.13-arch1-1`. RGB, brightness and fan control work well.

```
➜  TUF-Control git:(master) sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FX705DT.308
	Release Date: 09/19/2019
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FX705DT
	Version: 1.0
```

Syzygianinfern0, 2020-05-22:
@ATechnoHazard have you gotten the GPU Drivers to work well with Prime Offloading in this laptop on Arch Linux?

ATechnoHazard, 2020-05-22:
Nope, I'm just using the nvidia for everything.

hackbnw, 2021-02-21:
Thanks for feedback!

commit fedc6a89 (2021-02-21): Updates supported model list info for FX705DT, closes #37
closed 2021-02-21 by hackbnw