hackbnw/faustus #38: Support for FA506IV

linuxazo, 2020-05-24:
_sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3_
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA506IV.304
	Release Date: 03/24/2020
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA506IV
	Version: 1.0

[dsdt.zip](https://github.com/hackbnw/faustus/files/4673267/dsdt.zip)

Willing to test further and provide feedback.  Thank You.

luisbarrancos, 2020-06-02:
Adding an extra here, version 302 instead of 304 for the same model.

BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA506IV.302
        Release Date: 03/04/2020
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA506IV
        Version: 1.0

[dsdt.zip](https://github.com/hackbnw/faustus/files/4714530/dsdt.zip)

tromoto, 2020-06-13:
in the fork: https://github.com/tromoto/faustus  there is (at the moment a feeble support)  for all the 506xx/706xx models, but not really tested properly. On a 1st sight it seems that 706xx have a bit different interface.
Need more testing on ubuntu 20.04 (5.6 oem kernel) and Arch/Manjaro.

linuxazo, 2020-06-14:
I tested on current Arch Linux with 5.7.2 kernel, module built as dkms. 
- Keyboard brightness keys are working!  Fn+Arrow Up/Down 
- Aura keys (Fn+Arrow Left/Right) get registered by faustus driver as: 
_faustus: Unknown key b3 pressed
faustus: Unknown key b2 pressed_
- Fan control key (Fn+F5) does get registered and acknowledged by the faustus driver. 
It seems to be working correctly switching between normal/boost/silent modes.
There is no OSD notification though.  One has to verify through "dmesg | grep faustus" as follows: 
_faustus: Set throttle thermal policy mode: 1
faustus: Set throttle thermal policy mode: 2
faustus: Set throttle thermal policy mode: 0_
I have not been able to test properly the RGB keyboard colour switching or light cycles... 
The set_rgb.sh script does not have any useful instructions for use... 
Would been nice to be able to monitor fan speeds through lm_sensors interface or otherwise...
Willing to test further... 
Thank You so much for your efforts!

Output of dmesg: [    2.664247] faustus: DMI checK: FA506IV
_[    2.664411] faustus: Initialization: 0x1
[    2.664504] faustus: BIOS WMI version: 8.1
[    2.664592] faustus: SFUN value: 0x4a0061
[    2.664595] faustus faustus: Use DSTS
[    2.664596] faustus faustus: Enable event queue
[    2.665274] faustus: Fan boost mode check successful
[    2.665274] faustus: Fan boost mode check end
[    2.666532] faustus: TTP check successful
[    2.666533] faustus: TTP check end
[    2.666534] faustus: Set throttle thermal policy mode: 0
[    2.671827] input: Asus WMI hotkeys as /devices/platform/faustus/input/input6_

tromoto, 2020-06-14:
Big thx to you on checking it out. I'll try to do more work after I get second nvme and (re)install Arch and Tumbleweed. It is a bit tricky to do testing on 20.04 and unknown 5.6x oem kernel (no idea for which box is this made, Ryzen 4x00 from Lenovo or Dell, very unlikely that is for Asus ?)

linuxazo, 2020-06-14:
I am sorry but I would not know what configuration options are in the stock Ubuntu kernel... I guess it is generic more or less... 
Let me know how I can help... I could help with further testing gladly on my FA506IV. 
Right now I am on 5.7.2 Arch Zen kernel.

linuxazo, 2020-06-14:
Also, now I can see the cpu fan speed, albeit it reports 0 RPM for now.

linuxazo, 2020-06-15:
Good morning,

It would be good if we could have the ability to turn off/on the LED
keyboard, as in aura_settings.png attachment.
To disable LED effects while the system is BOOT UP, or SLEEP or SHUT DOWN
and just keep it on or turn the LED keyboard backlight off while the system
is AWAKE.

Thank You for consideration.

lvalladares, 2020-08-11:
Hello!

I've tested your version @tromoto in Linux Mint with Kernel 5.7 and 5.8, in my case the keyboard colors works greath, i can change the keyboard brightness using the hotkeys and using the tuf-control app i can change the color of the RGB.

But i cant change the thermal policy, when i try to do it and i check dmesg i get this:

```
[  708.032887] faustus: Set throttle thermal policy mode: 1
[  708.034505] faustus: Failed to set throttle thermal policy (retval): 0x1
[  709.225762] faustus: Set throttle thermal policy mode: 1
[  709.227462] faustus: Failed to set throttle thermal policy (retval): 0x1
[  710.165633] faustus: Set throttle thermal policy mode: 0
[  710.167365] faustus: Failed to set throttle thermal policy (retval): 0x1
[  711.268707] faustus: Set throttle thermal policy mode: 1
[  711.270869] faustus: Failed to set throttle thermal policy (retval): 0x1
[  724.248171] faustus: Set throttle thermal policy mode: 2
[  724.250191] faustus: Failed to set throttle thermal policy (retval): 0x1
```

Also (i think its not related) linux cant see my fan sensors

Frick-David, 2020-12-03:
Building on this I have this for dmidecode:
`
   
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA506IV.309
        Release Date: 07/02/2020
        BIOS Revision: 5.16
        Firmware Revision: 3.9

System Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF Gaming FA506IV_FA506IV
        Version: 1.0
        Family: TUF Gaming FA506IV
`

Can I help test this (without blowing up my laptop)?

zneix, 2021-02-18:
Arch Linux 5.10.16 user reporting in
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA506IV.313
        Release Date: 11/16/2020
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA506IV
        Version: 1.0
```

This repo didn't work for me, but [this fork](https://github.com/Hyper-KVM/faustus) does the job perfectly and I wonder if @Hyper-KVM wants to create a Pull Request to merge his changes to this upstream and make it also work on our laptop model.

amija004, 2021-04-10:
```
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
```

```
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA506IV.309
	Release Date: 07/02/2020
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA506IV
	Version: 1.0
```

I would second @zneix\'s request for a Pull Request, as @Hyper-KVM\'s fork works for me.