hackbnw/faustus #40: Question related to fan control

RadoiAndrei, 2020-06-27:
Not related to the project, but I love it's ideea.
I own a FX505DT and I want to know how can I know the fan preset I am on?
I know there are those 3 Silent/Normal/Overboost, but I never know on which one I am, any ideea how I can check that?

icodelifee, 2020-06-28:
https://github.com/hackbnw/faustus#fan-mode

RadoiAndrei, 2020-06-28:
> https://github.com/hackbnw/faustus#fan-mode

Yeah, I saw that, but I mean as in an indicator or anything visual.

RadoiAndrei, 2020-06-29:
Can we not implement some sort of hook in the driver so we can execute a script or something on fan control mode change?
I'd love to use `notify-send` when the mode changes to notify me with the new mode in use.

closed 2020-06-30 by RadoiAndrei