hackbnw/faustus #41: Works with FX505DT

RadoiAndrei, 2020-06-28:
**Output for `sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3`**
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505DT.310
        Release Date: 12/24/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505DT
        Version: 1.0
```
[image]

RadoiAndrei, 2020-06-28:
[image]

Also, this means I fixed the graphic cards issue, right? I am still quite a newbie when it comes to linux.

icodelifee, 2020-06-28:
is the display flickering for you?

RadoiAndrei, 2020-06-28:
> is the display flickering for you?
> 

I did not experience any flickering yet.

icodelifee, 2020-06-28:
its supposed to happen display goes to sleep and wakes up!
oh so you have different panel?
mine only has 48hz and 120hz

RadoiAndrei, 2020-06-28:
> its supposed to happen display goes to sleep and wakes up!
> oh so you have different panel?
> mine only has 48hz and 120hz

Nope, does not happen to me.

sticsk, 2020-07-04:
It is working for me, using [https://github.com/rublag/faustus](url) this repo, I have the same laptop as you

icodelifee, 2020-07-05:
> It is working for me, using [https://github.com/rublag/faustus](url) this repo, I have the same laptop as you

what is working for you in your repo?

sticsk, 2020-07-05:
> > It is working for me, using [https://github.com/rublag/faustus](url) this repo, I have the same laptop as you
> 
> what is working for you in your repo?

rgb lights works

Fan Works but it takes a bit of time to realise, and when i hit `dmesg -wH` it says `Failed to set throttle thermal policy`, but I did a benchmark test with Unigine Heaven and the fans were going off when on boost mode so it's a good sign, but I don't use the boost mode I use silent mode for battery, I'll check the clock speed later to see if Silent it's working.

I don't know what's modified in rublag's repo beacuse I don't understand where to check, but hackbnw's fan modes didnt work.

By the way my bios version is FX505DT.310 and kernel is linux57

icodelifee, 2020-07-05:
> rgb lights works
>
> Fan Works but it takes a bit of time to realise, and when i hit `dmesg -wH` it says `Failed to set throttle thermal policy`, but I did a benchmark test with Unigine Heaven and the [...]

and you have the same display as @RadoiAndrei ?

sticsk, 2020-07-05:
Yep, same gpu/cpu with non-free drivers

icodelifee, 2020-07-05:
> Yep, same gpu/cpu with non-free drivers

hmm, weird i guess only indian and some eu variants have display with 48hz and 120hz mode!

sticsk, 2020-07-05:
> > Yep, same gpu/cpu with non-free drivers
> 
> hmm, weird i guess only indian and some eu variants have display with 48hz and 120hz mode!
> 

The guy's Romanian and that version of the laptop comes with 60Hz, the 120Hz is expensive with i7 or the ryzen equivalent(i forgot)

ferretwithaberet, 2021-01-19:
It is no surprise, but it works on EndeavourOS too:
[image]

commit a51cb48f (2021-03-14): Update supported model list info for FX505DT, closes #41
closed 2021-03-14 by hackbnw