hackbnw/faustus #42: Hook for fan control mode change

RadoiAndrei, 2020-06-29:
I want to be able to run a script on fan control mode change. I do not know entirely how linux programming works yet, but I suppose you could let us do something like
```bash
export AFTER_FAN_MODE_CHANGE="./some-script.sh"
```
and actually have that script be run by the function that changes the fan control mode.

rublag, 2020-06-30:
I added sysfs_notify on my fork
https://github.com/rublag/faustus

So you can use poll to get throttle thermal policy mode

example python script:
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

import select

TTP_PATH = "/sys/devices/platform/faustus/throttle_thermal_policy"
ttp_fd = open(TTP_PATH, "r")

print("We need to do read first.")
print(f"TTP: {ttp_fd.read()}", end='')

print("Setting change event polling")
poller = select.poll()
poller.register(ttp_fd, select.POLLERR | select.POLLPRI)

print("Starting polling loop")
while True:
    events = poller.poll()
    print("Got events. Now we need to reopen or seek(0) fd, then read")

    ttp_fd.seek(0)
    print(f"TTP: {ttp_fd.read()}")
```

cromerc, 2020-07-29:
Using rublag's fork and example I came up with this python script:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import select
import subprocess

def sendmessage(message):
    print(message)
    subprocess.Popen(['notify-send', '-t', '3000', message])
    return

TTP_PATH = "/sys/devices/platform/faustus/throttle_thermal_policy"
ttp_fd = open(TTP_PATH, "r")

poller = select.poll()
poller.register(ttp_fd, select.POLLERR | select.POLLPRI)

ttp_fd.seek(0)
mode = int(ttp_fd.read())
if mode == 0:
    print("Fan is in normal mode")
elif mode == 1:
    print("Fan is in boost mode")
elif mode == 2:
    print("Fan is in silent mode")

while True:
    events = poller.poll()

    ttp_fd.seek(0)
    mode = int(ttp_fd.read())
    if mode == 0:
        sendmessage("Fan is in normal mode")
    elif mode == 1:
        sendmessage("Fan is in boost mode")
    elif mode == 2:
        sendmessage("Fan is in silent mode")
```

I then set this script to run in the background when my computer starts. It sends me a little popup window telling me what fan speed is set when I hit Fn+ F5

ferretwithaberet, 2020-08-26:
@rublag polling does not seem to work on my FX505DT. I should be able to test if it works if I run the python file and press `Fn + F5` right? It only prints the initial mode, and gets stuck in the `poller.poll()` but seems to never catch any events.

rublag, 2020-08-26:
@ferretwithaberet 
asus-nb-wmi driver doesn't support it. My faustus fork supports only for throttle thermal policy (not for fan boost mode). 
> I should be able to test if it works if I run the python file and press `Fn + F5` right?

Yes.
> It only prints the initial mode, and gets stuck in the `poller.poll()` but seems to never catch any events.

Because kernel has no way to know if something changed. For polling support driver has to call `sysfs_notify`.

ferretwithaberet, 2020-08-26:
> @ferretwithaberet
> asus-nb-wmi driver doesn't support it. My faustus fork supports only for throttle thermal policy (not for fan boost mode).
>
>
> Yes.
>
>
> Because kernel has no way to know if something [...]

The problem is that I got your fork installed, I installed it through AUR, and it does not work.

rublag, 2020-08-26:
@ferretwithaberet 
What does it print to kernel log when you press `Fn-F5` (check with `dmesg` command)?

ferretwithaberet, 2020-08-26:
@rublag I am a newbie, when I press `Fn-F5` I think it shows nothing related to it, I got this:
```
[  646.690654] kauditd_printk_skb: 15 callbacks suppressed
[  646.690657] audit: type=1101 audit(1598482883.769:362): pid=5350 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="keydev" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  646.691026] audit: type=1110 audit(1598482883.769:363): pid=5350 uid=0 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_unix,pam_permit,pam_env acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  646.696336] audit: type=1105 audit(1598482883.772:364): pid=5350 uid=0 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
```

rublag, 2020-08-26:
@ferretwithaberet 
It has to write some debug info. Run these commands and show output of each.
`lsmod | grep asus`
`lsmod | grep faustus`
`dmesg | grep throttle`
`dmesg | grep boost`

ferretwithaberet, 2020-08-26:
`lsmod | grep asus`
```
asus_wireless          20480  0
```

`lsmod | grep faustus`
```
faustus                45056  0
battery                20480  1 faustus
sparse_keymap          16384  1 faustus
rfkill                 28672  9 faustus,bluetooth,cfg80211
wmi                    36864  2 faustus,wmi_bmof
```

`dmesg | grep throttle` and `dmesg | grep boost` showed nothing. I am running the script you wrote above on my user, not on root.

rublag, 2020-08-26:
@cromerc 
Okay I've just looked in the PKGBUILD of the faustus-rublag-dkms-git AUR package.
It has error on line 14. Clones sources from hackbnw's repository, not mine.

@ferretwithaberet 
```
git clone https://github.com/rublag/faustus.git
cd faustus
make
sudo rmmod faustus
sudo insmod src/faustus.ko let_it_burn
```

ferretwithaberet, 2020-08-26:
> @cromerc
> Okay I've just looked in the PKGBUILD of the faustus-rublag-dkms-git AUR package.
> It has error on line 14. Clones sources from hackbnw's repository, not mine.
>
> @ferretwithaberet
>
> ```
> git [...]

@rublag Oh, that makes sense now why it was not working. Thank you, now it works.

cromerc, 2020-08-26:
Good catch, I will fix that.

cromerc, 2020-08-26:
Also, you can give tuf-manager a try from the AUR. It comes with a program that runs on the status bar and notifies you om fan mode changes.

ferretwithaberet, 2020-08-26:
@cromerc By the way, can you tell me how you set that script to run in the background using polling? Did you create a unit service, and if yes, how does the service file look, I want an example to base mine on.

cromerc, 2020-08-26:
@rublag @ferretwithaberet Fixed the AUR package url. Now it will work.

@ferretwithaberet The script I showed you I didn't actually use, I just wrote it as an example. The way I got the background service working in tuf-manager was by putting a desktop file in /etc/xdg/autostart that calls my program. Desktop environments such as cinnamon, mate, gnome, etc look in /etc/xdg/autostart and start the services when the user logs in.

closed 2020-08-31 by ferretwithaberet