hackbnw/faustus #44: Module won't load on 2020 17" TUF Laptop FA706IU - newly released unit, Ryzen 7 4800H

PhilSalkie, 2020-07-31:
Builds fine on kernel 5.7.0, but fails to find its desired device, even in "burn it down" mode.

Keyboard LEDs are in some sort of Wurlitzer Jukebox color-cycling mode, and nothing changes them (they do turn off after some inactivity in battery-only mode.)   I've tried OpenRGB, it can't find any devices to control.  This kernel has the patch to enable the second I2C bus, but that doesn't seem to have any Aura controllers on it.

I see "LED" mentioned a few times in a WMI section of the decompiled AML file, don't know if that's hope, or false hope.

I'm happy to do any experiments to help reverse-engineer this - at the moment, I've only got the Linux install, so I can't do any tracing of the Asus Windows tools, but I'll probably get to making a Windows 10 SSD soon.  Contact: philsalkie@gmail.com

Here's some info:

BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA706IU.309
	Release Date: 07/02/2020

Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA706IU
	Version: 1.0

AML file:  [http://asylumhouse.net/asus/dsdt.aml](url)

This is from the MX19 Linux info-grabber, may be helpful:

[code]

System:
  Host: tuftux Kernel: 5.7.0-2-amd64 x86_64 bits: 64 compiler: gcc v: 8.3.0 
  parameters: BOOT_IMAGE=/boot/vmlinuz-5.7.0-2-amd64 
  root=UUID=5a870cb6-5152-49e9-84ab-f2cc8cbc6cae ro quiet splash 
  nvme_core.default_ps_max_latency_us=5500 
  Desktop: Xfce 4.14.2 tk: Gtk 3.24.5 info: xfce4-panel wm: xfwm4 
  dm: LightDM 1.26.0 Distro: MX-19.2_ahs_x64 patito feo May 31  2020 
  base: Debian GNU/Linux 10 (buster) 
Machine:
  Type: Laptop System: ASUSTeK product: TUF Gaming FA706IU_TUF706IU v: 1.0 
  serial: <filter> 
  Mobo: ASUSTeK model: FA706IU v: 1.0 serial: <filter> 
  UEFI: American Megatrends v: FA706IU.309 date: 07/02/2020 
Battery:
  ID-1: BAT1 charge: 65.0 Wh condition: 91.9/90.2 Wh (102%) volts: 5.0/15.9 
  model: ASUS A32-K55 type: Li-ion serial: <filter> status: Discharging 
CPU:
  Topology: 8-Core model: AMD Ryzen 7 4800H with Radeon Graphics bits: 64 
  type: MT MCP arch: Zen family: 17 (23) model-id: 60 (96) stepping: 1 
  microcode: 8600104 L2 cache: 4096 KiB 
  flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm 
  bogomips: 92626 
  Speed: 1692 MHz min/max: 1400/2900 MHz boost: enabled Core speeds (MHz): 
  1: 1692 2: 1700 3: 1574 4: 1398 5: 1397 6: 1329 7: 1397 8: 1406 9: 1336 
  10: 1430 11: 1603 12: 1224 13: 1414 14: 1532 15: 1251 16: 1247 
  Vulnerabilities: Type: itlb_multihit status: Not affected 
  Type: l1tf status: Not affected 
  Type: mds status: Not affected 
  Type: meltdown status: Not affected 
  Type: spec_store_bypass 
  mitigation: Speculative Store Bypass disabled via prctl and seccomp 
  Type: spectre_v1 
  mitigation: usercopy/swapgs barriers and __user pointer sanitization 
  Type: spectre_v2 mitigation: Full AMD retpoline, IBPB: conditional, 
  IBRS_FW, STIBP: conditional, RSB filling 
  Type: srbds status: Not affected 
  Type: tsx_async_abort status: Not affected 
Graphics:
  Device-1: NVIDIA TU116M vendor: ASUSTeK driver: N/A bus ID: 01:00.0 
  chip ID: 10de:2191 
  Device-2: AMD Renoir vendor: ASUSTeK driver: amdgpu v: kernel 
  bus ID: 05:00.0 chip ID: 1002:1636 
  Display: x11 server: X.Org 1.20.4 driver: amdgpu,ati 
  unloaded: fbdev,modesetting,vesa resolution: 1920x1080~120Hz 
  OpenGL: renderer: AMD RENOIR (DRM 3.37.0 5.7.0-2-amd64 LLVM 10.0.0) 
  v: 4.6 Mesa 20.0.7 direct render: Yes 
Audio:
  Device-1: NVIDIA vendor: ASUSTeK driver: snd_hda_intel v: kernel 
  bus ID: 01:00.1 chip ID: 10de:1aeb 
  Device-2: AMD vendor: ASUSTeK driver: snd_hda_intel v: kernel 
  bus ID: 05:00.1 chip ID: 1002:1637 
  Device-3: AMD Raven/Raven2/FireFlight/Renoir Audio Processor 
  vendor: DekTec Digital Video B.V. driver: N/A bus ID: 05:00.5 
  chip ID: 1022:15e2 
  Device-4: AMD Family 17h HD Audio vendor: ASUSTeK driver: snd_hda_intel 
  v: kernel bus ID: 05:00.6 chip ID: 1022:15e3 
  Sound Server: ALSA v: k5.7.0-2-amd64 
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet 
  vendor: ASUSTeK driver: r8169 v: kernel port: e000 bus ID: 02:00.0 
  chip ID: 10ec:8168 
  IF: eth0 state: down mac: <filter> 
  Device-2: Realtek vendor: AzureWave driver: rtw_pci v: N/A port: d000 
  bus ID: 03:00.0 chip ID: 10ec:c822 
  IF: wlan0 state: up mac: <filter> 
Drives:
  Local Storage: total: 1.86 TiB used: 760.99 GiB (39.9%) 
  ID-1: /dev/nvme0n1 model: PCIe SSD size: 1.86 TiB block size: 
  physical: 512 B logical: 512 B speed: 31.6 Gb/s lanes: 4 serial: <filter> 
  rev: ECFM22.6 scheme: GPT 
Partition:
  ID-1: / raw size: 1.83 TiB size: 1.80 TiB (98.38%) 
  used: 760.97 GiB (41.3%) fs: ext4 dev: /dev/nvme0n1p2 
  ID-2: swap-1 size: 34.03 GiB used: 24.4 MiB (0.1%) fs: swap 
  swappiness: 15 (default 60) cache pressure: 100 (default) 
  dev: /dev/nvme0n1p3 
Sensors:
  System Temperatures: cpu: 33.8 C mobo: N/A gpu: amdgpu temp: 33 C 
  Fan Speeds (RPM): N/A 
Repos:
  No active apt repos in: /etc/apt/sources.list 
  Active apt repos in: /etc/apt/sources.list.d/debian-stable-updates.list 
  1: deb http://deb.debian.org/debian buster-updates main contrib non-free
  Active apt repos in: /etc/apt/sources.list.d/debian.list 
  1: deb http://deb.debian.org/debian buster main contrib non-free
  2: deb http://deb.debian.org/debian-security buster/updates main contrib non-free
  3: deb http://deb.debian.org/debian buster-backports main contrib non-free
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list 
  1: deb [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/insync.list 
  1: deb http://apt.insync.io/debian buster non-free contrib
  Active apt repos in: /etc/apt/sources.list.d/mx.list 
  1: deb http://mirrors.rit.edu/mxlinux/mx-packages/mx/repo/ buster main non-free
  2: deb http://mirrors.rit.edu/mxlinux/mx-packages/mx/repo/ buster ahs
  Active apt repos in: /etc/apt/sources.list.d/skype-stable.list 
  1: deb [arch=amd64] https://repo.skype.com/deb stable main
  Active apt repos in: /etc/apt/sources.list.d/teamviewer.list 
  1: deb http://linux.teamviewer.com/deb stable main
  No active apt repos in: /etc/apt/sources.list.d/various.list 
Info:
  Processes: 412 Uptime: 2h 45m Memory: 30.86 GiB used: 2.22 GiB (7.2%) 
  Init: SysVinit v: 2.93 runlevel: 5 default: 5 Compilers: gcc: 8.3.0 alt: 8 
  Shell: bash v: 5.0.3 running in: quick-system-in inxi: 3.0.36 
[/code]

cromerc, 2020-07-31:
This fork supposedly works with FA706II, so it might work with FA706IU as well: https://github.com/rublag/faustus

This is the commit that added FA706 support: https://github.com/rublag/faustus/commit/be0885d8ea2cfb15dca94cfe1171649927b3572c

PhilSalkie, 2020-07-31:
It works!  (I haven't tried the fan controls, but the LEDs work, and
that's honestly what I came here for...)

OpenRGB sees it just fine, I added

options faustus let_it_burn=1

to /etc/modprobe.d/faustus.conf to get the module to load at boot.

Thank you so much for the help!

PhilSalkie, 2020-08-04:
Gave the fan controls a try, it's the same as the previous issue - dmesg just says "Failed to set throttle thermal policy (retval): 0x1"

hackbnw, 2021-02-28:
Merged the mainline back into driver. The specific issue was fixed in 

https://github.com/torvalds/linux/commit/7b91f1565fbfb

Notice however this likely means that the RGB API might have been changed as well. Haven't yet time to look into it.

hackbnw, 2021-03-28:
Thanks for the feedback, the specific bug in question should be fixed, see https://github.com/hackbnw/faustus/issues/62#issuecomment-803635815 . I'm going to close the issue.

closed 2021-03-28 by hackbnw