hackbnw/faustus #46: New UI

cromerc, 2020-08-05:
I wrote a new UI for this driver module: [https://git.cromer.cl/cromer/tuf-manager](https://git.cromer.cl/cromer/tuf-manager)

It would be nice to have it linked from the readme file here.

The software includes 2 different user interfaces, CLI and GUI. It is written in Vala and uses GTK3 for the GUI. I am still working on it but it is usable now.

hackbnw, 2021-03-21:
Hi, thanks! I've included the link to your project in UI section.

closed 2021-03-21 by hackbnw

cromerc, 2021-03-21:
Considering the changes you have been doing recently, the kernel should notify on fan mode change now?

If so I will update the project to point to this branch of development.

hackbnw, 2021-03-21:
The code here is almost 1-to-1 original asus-wmi + RGB. I've recently merged the mainline from torvalds/linux in 707f55ba87256ab3ae93b9d0d57bd9bf4051115c, which did include patches by rublag. It should do sysfs_notify now, as are all recent kernels. As he pointed out in #47 the end goal here is to put everything in mainline eventually.

hackbnw, 2021-03-21:
The fan issue workaround is not there though.

commit b5399fd2 (2021-03-21): Add link to tuf-manager UI, closes #46