hackbnw/faustus #48: Getting "No Such Device" Error on F506

Karan96Kaushik, 2020-08-07:
Need help,

**BIOS Information**
        Vendor: American Megatrends Inc.
        Version: FX506LH.304
**System Information**
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF Gaming FX506LH_FX566LH

Following are the commands that I run:
git clone <faustus>
make

**Errors:**
1. I get "
[ 1122.173987] sparse_keymap: unknown parameter 'wmi' ignored
[ 1122.173991] sparse_keymap: unknown parameter 'video' ignored
" in dmesg
after running "sudo modprobe sparse-keymap wmi video"
2. rmmod: ERROR: Module asus_wmi is not currently loaded
3. rmmod: ERROR: Module asus_nb_wmi is not currently loaded
4. insmod: ERROR: could not insert module src/faustus.ko: **No such device**

cromerc, 2020-08-08:
Did you disable the original modules? You have to do this like it says in the readme:
Create file /etc/modprobe.d/faustus.conf with following contents
blacklist asus_wmi
blacklist asus_nb_wmi

Make sure to reboot after you make that file for it to take effect.

cromerc, 2020-08-08:
Also looking at the command you ran, you didn't do what it says in the readme either...

You have to do:
make
sudo modprobe sparse-keymap wmi video
sudo insmod src/faustus.ko

Please follow the direction in the readme.

Karan96Kaushik, 2020-08-09:
Yes, I tried exactly as mentioned in the Readme, I'm getting the same errors. As I mentioned, "sudo modprobe sparse-keymap wmi video" gives error in dmesg

cromerc, 2020-08-09:
Are you using a custom kernel? What distro are you on?

Karan96Kaushik, 2020-08-09:
> Are you using a custom kernel? What distro are you on?

Ubunutu 20.04

cromerc, 2020-08-09:
Do you have "linux-modules-extra" package installed?

Karan96Kaushik, 2020-08-09:
> Do you have "linux-modules-extra" package installed?

Yes, I have (linux-modules-extra-5.4.0-42-generic)

syfluqs, 2020-09-18:
you can try insmod with the `let_it_burn` parameter, but no such device means that the hardware device that the driver is looking for does not exist. `let_it_burn=1` will bypass the hardware DMI check. I have a FX566IH and I also am getting the same error without `let_it_burn`. If I bypass the DMI check, only the features in the mainline kernel (from `asus_nb_wmi`) work, so only changing brightness of the keyboard leds but not color. Clearly this kernel module needs a patch for the newer ASUS Tuf versions (FX506xx/FX566xx).

cromerc, 2020-09-18:
You could try this fork: https://github.com/rublag/faustus

It added support for some more models that this project doesn't yet support. No guarantees it will work for your models though, but worth a shot to try it.

syfluqs, 2020-09-18:
Upon further searching, https://github.com/tromoto/faustus works without a manual DMI check override for me, although I got `faustus: Failed to set throttle thermal policy (retval): 0x1` in `dmesg`. Any idea what this could mean?

hackbnw, 2021-02-21:
```
[ 1122.173987] sparse_keymap: unknown parameter 'wmi' ignored
[ 1122.173991] sparse_keymap: unknown parameter 'video' ignored
```

The `modprobe` syntax was wrong, I've corrected it in README, and added a section about "No such device" error.

commit 6eceaa6e (2021-02-21): Update README with info about DMI check and correct modprobe syntax, closes #48
closed 2021-02-21 by hackbnw