hackbnw/faustus #49: Works with FX505GT

teodorzl, 2020-08-07:
Works with FX505GT, BIOS version FX505GT.305, Ubuntu 20.04, kernel version 5.4.0-42-generic.
Tested and everything works as intended.

Rotanish, 2020-08-08:
I confirm. It works on my FX505GT. BIOS version FX505GT.304, OS: Manjaro 20.0.3 Lysia, Kernel: x86_64 Linux 5.4.52-1-MANJARO
Thanks for the work done, it's amazing.
I added `options faustus let_it_burn = 1` to `/etc/modprobe.d/faustus.conf` for automatic loading.

But OpenRGB doesn't see keyboard backlighting.

How to set up a notification when change fan mode?

```
> sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505GT.304
        Release Date: 11/11/2019
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505GT
        Version: 1.0
```

[dstd.aml.tar.gz](https://github.com/hackbnw/faustus/files/5044729/dstd.aml.tar.gz)

cromerc, 2020-08-08:
If you want notifications on fan changes check out [TUF Manager](https://git.cromer.cl/cromer/tuf-manager). It has a tray icon that runs after you login and notifies on fan mode change. It does require the use of the fork of faustus here: https://github.com/rublag/faustus

The reason it requires the fork is because the fork allows polling of the fan mode. TUF Manager takes advantage of this to do notifications.

Other benefits of TUF Manager include:
- Restore fan mode after reboot
- Different Keyboard lighting for different users/distros
- GUI control for fans and keyboard rgb lighting
- CLI interface for those that don't like GUI
- Support for 4 init systems, OpenRC, runit, s6, and systemd.

naryal2580, 2020-10-02:
It's working all good, with my **FX505GT**
```
❯ uname -a
Linux tuf 5.4.64-1-MANJARO #1 SMP PREEMPT Wed Sep 9 18:26:19 UTC 2020 x86_64 GNU/Linux

❯ sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FX505GT.305
	Release Date: 01/08/2020
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FX505GT
	Version: 1.0
```
[dsdt.aml.tar.gz](https://github.com/hackbnw/faustus/files/5317141/dsdt.aml.tar.gz)

@hackbnw please do add FX505GT in the support table, and get us rid of the `let_it_burn=1` thing :sweat_smile:

commit a08e6005 (2021-02-21): Add FX505GT to supported models list, closes #49
closed 2021-02-21 by hackbnw