hackbnw/faustus #54: Tested on FA706II (Failed)

BohdanNezhyvyi, 2020-09-21:
uname -a
`Linux ASUS-TUF-A17 5.8.6-1-MANJARO #1 SMP PREEMPT Thu Sep 3 14:19:36 UTC 2020 x86_64 GNU/Linux`

sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
```
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA706II.309
        Release Date: 07/02/2020

Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA706II
        Version: 1.0
```

[dsdt.aml.zip](https://github.com/hackbnw/faustus/files/5258250/dsdt.aml.zip)

after installing via make dkms, blaklisting asus_wmi, asus_nb_wmi, and sudo insmod ./src/faustus.ko let_it_burn=1 got this message:

`insmod: ERROR: could not insert module ./src/faustus.ko: No such device`

dmesg say:

```
[ 1566.301464] faustus: Omitting DMI verification
[ 1566.301660] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200528/dsopcode-198)
[ 1566.301670] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200528/psparse-529)
[ 1566.301727] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200528/dsopcode-198)
[ 1566.301733] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200528/psparse-529)
[ 1566.301786] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200528/dsopcode-198)
[ 1566.301791] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200528/psparse-529)
[ 1566.301803] faustus faustus: Use DSTS
[ 1566.301805] faustus faustus: Enable event queue
[ 1566.301901] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200528/dsopcode-198)
[ 1566.301906] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200528/psparse-529)
[ 1566.301957] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Field [IIA3] at bit offset/length 96/32 exceeds size of target Buffer (96 bits) (20200528/dsopcode-198)
[ 1566.301962] ACPI Error: Aborting method \_SB.ATKD.WMNB due to previous error (AE_AML_BUFFER_LIMIT) (20200528/psparse-529)
[ 1566.301976] faustus: probe of faustus failed with error -5
[ 1566.301998] faustus: Can't probe platform driver: -19
```

cromerc, 2020-09-22:
FA706 works with this fork: https://github.com/rublag/faustus

BohdanNezhyvyi, 2020-09-22:
> FA706 works with this fork: https://github.com/rublag/faustus

I already tried this fork, doesn't work

oshnix, 2020-10-15:
I've tried to run with the fork, but got the following error:

```
[ 2924.600696] faustus: Omitting DMI verification
[ 2924.600805] faustus: Initialization: 0x1
[ 2924.600833] faustus: BIOS WMI version: 8.1
[ 2924.600856] faustus: SFUN value: 0x4a0061
[ 2924.600857] faustus faustus: Use DSTS
[ 2924.600858] faustus faustus: Enable event queue
[ 2924.601096] faustus: Fan boost mode check successful
[ 2924.601097] faustus: Fan boost mode check end
[ 2924.601880] faustus: TTP check successful
[ 2924.601881] faustus: TTP check end
[ 2924.601883] faustus: Set throttle thermal policy mode: 0
[ 2924.602528] faustus: Failed to set throttle thermal policy (retval): 0x1
[ 2924.604582] input: Asus WMI hotkeys as /devices/platform/faustus/input/input27
[ 2924.604969] faustus: asus_wmi_fan_init start
[ 2924.606125] faustus: It seems that newer models don't have verification for method
[ 2924.610736] battery: new extension: ASUS Battery Extension
```

aalovasz, 2020-11-16:
Hi, I'm getting the same error on my FA706II, running Pop OS (new to linux, not sure what else can I mention), but I'm usually good at following steps...

> insmod: ERROR: could not insert module ./src/faustus.ko: No such device

kean83, 2020-11-23:
Same error here on POP 20.10 / 5.9.8 :(

BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA706IU.309
	Release Date: 07/02/2020
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA706IU
	Version: 1.0

hackbnw, 2021-02-28:
Merged the mainline back into driver. The specific issue was fixed in 

https://github.com/torvalds/linux/commit/7b91f1565fbfb

Notice however this likely means that the RGB API might have been changed as well. Haven't yet time to look into it.

hackbnw, 2021-02-28:
See also #44

closed 2021-03-21 by hackbnw

hackbnw, 2021-03-21:
RGB & module loading should work now, see #62, I closed this issue, the fan issue will be tracked there.