hackbnw/faustus #55: Tried on Model FX505GT, BIOS FX505GT.308 [Failed]

muhdlaziem, 2020-11-11:
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX505GT.308
        Release Date: 09/22/2020
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX505GT
        Version: 1.0
Followed all the steps in readme  and tried let_it_burn cause the "No such device" message appear.

sudo insmod ./src/faustus.ko let_it_burn=1

dmesg says:
[  727.877550] faustus: SFUN value: 0x4a0061
[  727.877552] faustus faustus: Use DSTS
[  727.877554] faustus faustus: Enable event queue
[  727.944665] input: Asus WMI hotkeys as /devices/platform/faustus/input/input28
[  728.011172] battery: new extension: ASUS Battery Extension
[  728.011743] audit: type=1106 audit(1605134471.479:169): pid=79810 uid=0 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  728.011853] audit: type=1104 audit(1605134471.479:170): pid=79810 uid=0 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  731.352816] audit: type=1101 audit(1605134474.819:171): pid=80362 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_permit,pam_time acct="muhdlaziem" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  731.353009] audit: type=1110 audit(1605134474.819:172): pid=80362 uid=0 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  731.355083] audit: type=1105 audit(1605134474.823:173): pid=80362 uid=0 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'

naryal2580, 2020-12-14:
Did you install headers? `sudo pacman -S linux-latest-headers`

hackbnw, 2021-06-02:
Where do you get "no such device"? dmesg output looks ok to me (the auditd logs are not relevant)