hackbnw/faustus #58: Would it be worth trying on ASUS FX504GE

ShrirajHegde, 2021-02-02:
Would it be worth trying on ASUS FX504GE, the keyboard layout is similar but doesn't have RGB

hackbnw, 2021-02-21:
Hi @ShrirajHegde,

thanks for asking. If you do not have RGB, than you do not need the driver, because this is nearly the only commonly useful feature that a) different from the mainline, b) is not in the mainline kernel. The kernel on your screenshot from other issue (5.8) should likely work fine out of the box. Other feature being reporting fan mode change using the keyboard to the UI with additional flag.

closed 2021-02-21 by hackbnw

ShrirajHegde, 2021-02-22:
> Hi @ShrirajHegde,
>
> thanks for asking. If you do not have RGB, than you do not need the driver, because this is nearly the only commonly useful feature that a) different from the mainline, b) is not [...]

ok Today I learned, the fan mode works even though there's no visual feedback, any way to check what fan mode is on?? because this is blind toggling

What about the present button? Anyway to utilize that button? That doesn't do anything

cromerc, 2021-02-22:
If the rublag fork works for you, you can use tuf-manager: https://git.cromer.cl/cromer/tuf-manager

Tuf-manager has a tray icon that notifies you when the fan mode changes. But currently only works with the rublag fork: https://github.com/rublag/faustus unless the kernel polling gets ported here from that fork in the future.

hackbnw, 2021-02-28:
Regarding the mute button it should have been forwarded as can be easily verified by evtest program. Whether or not it is functional is the question of support in the desktop manager and other software:

```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
...
/dev/input/event23:	Asus WMI hotkeys
Select the device event number [0-23]: 23
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Asus WMI hotkeys"
...
Testing ... (interrupt to exit)
Event: time 1614525523.151239, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7c
Event: time 1614525523.151239, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1614525523.151239, -------------- SYN_REPORT ------------
Event: time 1614525523.151286, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1614525523.151286, -------------- SYN_REPORT ------------
```

@cromerc he did submit the sysfs notifications to the mainline https://github.com/torvalds/linux/commit/72ceec58685f8a5c2ec22d9617c7b1e239e2e809

ShrirajHegde, 2021-03-01:
> Regarding the mute button it should have been forwarded as can be easily verified by evtest program. Whether or not it is functional is the question of support in the desktop manager and other software:
> 
> ```
> $ sudo evtest
> No device specified, trying to scan all of /dev/input/event*
> Available devices:
> ...
> /dev/input/event23:	Asus WMI hotkeys
> Select the device event number [0-23]: 23
> Input driver version is 1.0.1
> Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
> Input device name: "Asus WMI hotkeys"
> ...
> Testing ... (interrupt to exit)
> Event: time 1614525523.151239, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7c
> Event: time 1614525523.151239, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
> Event: time 1614525523.151239, -------------- SYN_REPORT ------------
> Event: time 1614525523.151286, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
> Event: time 1614525523.151286, -------------- SYN_REPORT ------------
> ```
> 
> @cromerc he did submit the sysfs notifications to the mainline [torvalds/linux@72ceec5](https://github.com/torvalds/linux/commit/72ceec58685f8a5c2ec22d9617c7b1e239e2e809)

This ACPI event fixes it in a "hacky" way(only in Ubuntu, SELinux creates some problems in fedora), this mutes the mic but doesn't work in VMs, it would be great if it worked natively. I am a fedora user so let me check 5.11 and see what happens.
```
event=button/micmute MICMUTE 00000080 00000000 K
action=/usr/bin/amixer sset 'Capture',0 toggle
```