hackbnw/faustus #61: Question on finding WMI device addresses

JafarAkhondali, 2021-03-05:
Hi,
May I know how did you find these device addresses?
```
#define ASUS_WMI_DEVID_KBD_RGB		0x00100056
#define ASUS_WMI_DEVID_KBD_RGB2		0x00100057
```

hackbnw, 2021-03-07:
Hi, trial and error mostly. It can be guessed by examining DSDT very carefully, as they are taking a very specific pattern of arguments consistent with RGB.

closed 2021-03-07 by hackbnw

JafarAkhondali, 2021-05-04:
@hackbnw I've been able to find GUID and the WMI method, but still have no idea how to extract(or find?) the DEV ID address. Could you please give me some hints?
the method is called WMBH with GUID of "7A4DDFE7-5B5D-40B4-8595-4408E0CC7F56"

[acer-gamining-functions-wmi.txt](https://github.com/hackbnw/faustus/files/6419450/acer-gamining-functions-wmi.txt)

JafarAkhondali, 2021-05-05:
@hackbnw Nevermind, found it.