hackbnw/faustus #62: Works on FA706II

siredmar, 2021-03-08:
FA706II: installed with `let_it_burn=1`. Used commit `707f55ba87256ab3ae93b9d0d57bd9bf4051115c`

```
sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
[sudo] password for armin: 
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA706II.304
	Release Date: 03/24/2020
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA706II
	Version: 1.0
```

LED control working with vanilla OpenRGB Release 0.5 (https://gitlab.com/CalcProgrammer1/OpenRGB/-/releases)
I don't know how to test the FAN control.

cromerc, 2021-03-08:
You can use TUF-Manager to test and control the fan: https://git.cromer.cl/cromer/tuf-manager

siredmar, 2021-03-08:
I can not verify that the FAN control is working using tuf-manager. I hear no difference between `silent` and `turbo` mode.
However the keyboard backlight is working with tuf-manager and OpenRGB.

hackbnw, 2021-03-21:
Hey, I looked into the DSDT in #54 (FA706II.309), the set fan boost mode function is there but not implemented. There is 315 BIOS update available for this device https://www.asus.com/us/Laptops/For-Gaming/TUF-Gaming/ASUS-TUF-Gaming-A17/helpdesk_Download/ Can you please try updating and check if the issue persists?

kean83, 2021-03-21:
I just tested it and the RGB keyboard is now working.

BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA706IU.315
	Release Date: 01/29/2021
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA706IU
	Version: 1.0

Kernel: 5.11.8-051108-generic

I had to use let_it_burn=1 and I also added it to the faustus.conf as:
options faustus let_it_burn=1

hackbnw, 2021-03-21:
@kean83 does Fn-F5 work with this version? You should see `faustus: Set fan boost mode: ..` in dmesg and it should noticeably change fan sound almost immediately.

kean83, 2021-03-21:
Fn + F5 does not work and has no trace in dmesg. 
Also the Fn + L.Arrow/R.Arrows Does not work (Aura)
[ 7360.836048] faustus: Unknown key b2 pressed
[ 7361.316084] faustus: Unknown key b3 pressed

hackbnw, 2021-03-28:
Thanks for the feedback, I've added both models to the list and DMI table. The issues with fan mode and RGB keys are also now in README.

commit 02c4732d (2021-03-28): Add FA706II and FA706IU to supported models list, closes #62
closed 2021-03-28 by hackbnw