hackbnw/faustus #63: Working on FX506, details inside

dsm-iv-tr, 2021-03-25:
Hi there, thanks for your great work!

I've just compiled and tested the module on the "ASUS TUF Gaming F15" -> "TUF506L". It is labelled as having "Aura RGB" control, and is an Intel Core i7 + nVidia GTX 1650Ti machine. I believe the identifier is "FX506", possibly "FX506LI".

On testing, the set_rgb.sh script does indeed set the keyboard colour. Fan control does not work with Fn-F5, nor via echoing values to the sysfs nodes. Faustus' thermal control did not appear to work, either. Out of caution, I set my keyboard LEDs to white and rebooted, allowing the original WMI modules to reload. All seems well, and while my goal was simply to make the LEDs white again, I would be pleased to provide any debug information or testing you'd like to make this model fully supported.

Cheers.

hackbnw, 2021-03-28:
Hello,

thanks for testing, can you please provide output of

```
$ sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3 
```

so that I can add your model to the list.

dsm-iv-tr, 2021-03-28:
No problem. Here you are.

`$ sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX506LI.304
        Release Date: 06/05/2020
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX506LI
        Version: 1.0`

hackbnw, 2021-03-28:
Thank you! I've added your model to the list so it should load without explicit parameter.

Can you please also check if you can switch fan mode with Fn-F5 (see output of dmesg to check if it does anything). The fan sound should noticeably change if it did work. I would also guess that Fn-Left, Fn-Right are not bound too (as in https://github.com/hackbnw/faustus/issues/62#issuecomment-803671518) ?

dsm-iv-tr, 2021-03-28:
I've pulled the latest commits. Here's dmesg after initially loading the module.

```
[  631.053952] faustus: DMI check: FX506LI
[  631.054207] faustus: Initialization: 0x1
[  631.054289] faustus: BIOS WMI version: 8.1
[  631.054344] faustus: SFUN value: 0x4a0061
[  631.054345] faustus faustus: Use DSTS
[  631.054346] faustus faustus: Enable event queue
[  631.060597] input: Asus WMI hotkeys as /devices/platform/faustus/input/input26
[  631.072280] battery: new extension: ASUS Battery Extension
```

Fn-F5 does not change dmesg, and no fan speed changes are observed.

Fn-LEFT reports:

`[  834.438942] faustus: Unknown key b2 pressed`

Fn-RIGHT reports:
`
[  873.338599] faustus: Unknown key b3 pressed`

dsm-iv-tr, 2021-03-28:
Some more information for you:

[dsdt.aml.txt](https://github.com/hackbnw/faustus/files/6217555/dsdt.aml.txt)

For fan/thermal stuff, doing things like:

`echo 2 > throttle_thermal_policy`

does not return an error, and it does set the value, but nothing happens in hardware for any combination of throttle_thermal_policy or fan_boost_mode.

dsm-iv-tr, 2021-03-28:
Further exploration reveals that 0xC5 might be the speed control register for the fan on this machine. Under load, it moves from 25 to A7 and above, and under no load, it decreases. The fan speed also correlates with the changes in the EC register.

dsm-iv-tr, 2021-04-14:
Here is a very confused graph of the registers on my system, generated by ec-probe.

I suspect that 0x58 is temperature, and 0xC5 is the fan control register from empirical testing with `yes` and my ears.

I tried writing to 0xC5 (after loading the ec_sys module with write_support=y), but it appears to be read-only.

[image]

hackbnw, 2021-04-18:
Cool, thanks, I didn't know the trick with ec_sys module. It does look very useful. The 0x58 should indeed be temperature, which should already exposed as thermal zone. Looking through your DSDT, there is what looks like the EC registers definitions. This looked different in FX series.

```
OperationRegion (ERAM, SystemMemory, 0xFE708300, 0x0100)
Field (ERAM, AnyAcc, NoLock, Preserve)
{
    ..
    Offset (0x58), 
    CTMP,   8, 
    FAND,   1, 
    SDNT,   1, 
    VGAT,   1, 
        ,   1, 
    F2FT,   1, 
    S3LW,   1, 
    BTS4,   1, 
    TRT3,   1, 
    ...
    Offset (0x63), 
    DBAT,   16, 
    CUTE,   8, 
    FRMS,   16, 
    FRS2,   16, 
    MDCP,   16, 
    MBCP,   16, 
    ...
    Offset (0xC5), 
    VRTT,   8, 
    KBBL,   8, 
    DPSF,   1, 
    ...
}
```

The CTMP (core/cpu temperature?) seems to be exposed via thermal zone. VRTT is not to be found anywhere else (voltage regulator test tool?). FRMS (if I read this correctly it is a word register over 0x66 .. 0x67) is used in couple of places, notably also near the previously used GFBM SFBM (get / set fan boost mode) functions (which are empty in your model).

```
ThermalZone (THRM)
{
    ...
    Method (_TMP, 0, NotSerialized)  // _TMP: Temperature
    {
        ...
            Local1 = \_SB.PCI0.LPCB.EC0.CTMP
            TTMP = ((Local1 * 0x0A) + TBSE) /* \_TZ_.TBSE */
            ...
            Return (TTMP) /* \_TZ_.TTMP */
        ...
    }
    ...
}
```

what I noticed scrolling down, there is the following in the same thermal zone:

```
            Method (_SCP, 1, NotSerialized)  // _SCP: Set Cooling Policy
            {
                CTYP = Arg0
                If ((Arg0 == Zero))
                {
                    Debug = "THERM: _SCP(Active)"
                }
                Else
                {
                    Debug = "THERM: _SCP(Passive)"
                }
            }
```

This effectively did nothing in FX series previously, now it sets some external register. This is one lead (haven't yet check if there is Linux implementation for this). Another lead is your finding with 0x67, the FRMS is used near previously implemented SFBM GFBM calls (0x00110013 .. 0x00110015 near 0x00110018). I was looking into SFSS / 0x00110015 and variables used there and I've accidentally found this block:

```
Method (_QDD, 0, NotSerialized)  // _Qxx: EC Query, xx=0x00-0xFF
            {
                If (ECOK)
                {
                    Acquire (MUT0, 0x1000)
                    If ((FNGM == Zero))
                    {
                        DNOT (0xD1)
                        ODV0 = Zero
                        Notify (B0D4, 0x83) // Device-Specific Change
                        Notify (IETM, 0x85) // Device-Specific
                        Notify (IETM, 0x88) // Device-Specific
                    }
                    ElseIf ((FNGM == 0x02))
                    {
                        DNOT (0xD1)
                        ODV0 = One
                        Notify (B0D4, 0x83) // Device-Specific Change
                        Notify (IETM, 0x85) // Device-Specific
                        Notify (IETM, 0x88) // Device-Specific
                    }
                    ElseIf ((FNGM == One))
                    {
                        DNOT (0xD1)
                        ODV0 = 0x02
                        Notify (B0D4, 0x83) // Device-Specific Change
                        Notify (IETM, 0x85) // Device-Specific
                        Notify (IETM, 0x88) // Device-Specific
                    }

                    Release (MUT0)
                    ^^^PEG0.PEGP.NLIM = One
                    Notify (^^^PEG0.PEGP, 0xC0) // Hardware-Specific
                }
            }
```

the IETM, 0x88 call was previously used on FX in fan boost mode switching, compare with my laptop

```
Method (SFBM, 1, NotSerialized)
        {
            If ((Arg0 == Zero))
            {
                FANM = Zero
                ODV0 = Zero
                \_SB.PCI0.LPCB.EC0.PTPT ()
                Notify (\_SB.IETM, 0x88) // Device-Specific
                If (\_SB.PCI0.LPCB.EC0.ACPS ())
                {
                    Notify (\_SB.PCI0.PEG0.PEGP, 0xD1) // Hardware-Specific
                }
            }

            If ((Arg0 == One))
            {
                FANM = One
                ODV0 = One
                \_SB.PCI0.LPCB.EC0.PTP1 ()
                Notify (\_SB.IETM, 0x88) // Device-Specific
                If (\_SB.PCI0.LPCB.EC0.ACPS ())
                {
                    Notify (\_SB.PCI0.PEG0.PEGP, 0xD1) // Hardware-Specific
                }
            }

            If ((Arg0 == 0x02))
            {
                FANM = 0x02
                ODV0 = 0x02
                \_SB.PCI0.LPCB.EC0.PTP2 ()
                Notify (\_SB.IETM, 0x88) // Device-Specific
                If (\_SB.PCI0.LPCB.EC0.ACPS ())
                {
                    Notify (\_SB.PCI0.PEG0.PEGP, 0xD4) // Hardware-Specific
                }
            }
        }
```

Now that looks really promising. You could try calling DEVS on device 0x00110015 and see what happens, but I can't understand implementation of SFSS yet or how and where the _QDD is called (if ever).

hackbnw, 2021-04-18:
PS: FNGM should be 0x6

```
                Offset (0x06), 
                FNGM,   8, 
```

dsm-iv-tr, 2021-04-18:
That was a great find.

I'm a programmer (C/Java/various scripting languages) and I have a general idea of what I'm poking around with, but I don't know how to make those calls - when you say, "call DEVS on device 0x00110015", is DEVS an offset containing a pointer to some kind of function taking the device as parameter - eg. something like `Notify (DEVS, 0x00110015)`?

My ACPI knowledge is very limited, but I am willing to compile, modify and test to see what I can get. 

Cheers.

dsm-iv-tr, 2021-04-19:
I've loaded the `acpi_call` module (https://github.com/mkottman/acpi_call), and am poking around. Keeping a log for you.

```
root@Laptop:/proc/acpi# echo '\SFSS 0x00' > call
root@Laptop:/proc/acpi# cat call
0xffalled
--
root@Laptop:/proc/acpi# echo '\SFSS 0x01' > call
root@Laptop:/proc/acpi# cat call
0xffalled
--
root@Laptop:/proc/acpi# echo '\SFSS 0x02' > call
root@Laptop:/proc/acpi# cat call
0xffalled
--
root@Laptop:/proc/acpi# echo '\SFBM 0x00' > call
root@Laptop:/proc/acpi# cat call
0x0called
--
root@Laptop:/proc/acpi# echo '\SFBM 0x01' > call
root@Laptop:/proc/acpi# cat call
0x0called
--
root@Laptop:/proc/acpi# echo '\SFBM 0x02' > call
root@Laptop:/proc/acpi# cat call
0xffffffff
--
root@Laptop:/proc/acpi# echo '\GFBM' > call
root@Laptop:/proc/acpi# cat call
0x10003ed
```

hackbnw, 2021-04-20:
That sounds great. Cool find with the `acpi_call` module too. It seemed to have worked :smile:  I will try digging some more. Do you know what should it do on Windows? Can you switch fan modes? Is it 3 modes or standard active / passive of energy savings settings? What exactly are we looking for? I couldn't understand the meaning of SFSS, but it seems in fact to be reading some sensor (FRMS) if the mode (FNGM - fan gain mode?) is 0 or, I don't know, it might be broken or decoded incorrectly. I guess the first letter is not "Set" (maybe fan speed sensor?). The SFBM is empty on your device, what you got is ok (wasn't in FX series).

The code I posted is from the DSTS. You can look it up yourself by just executing `iasl ..aml` from `acpica-tools` and checking out the `..dsl` code (see the Ubuntu link below about WMI). The DSTS essentially describes how ACPI is implemented providing interface to BIOS, it's rather C like, but not quite (it's big though and not that well readable, you might want to search for the hex values / method names in text editor to skip to the part). The WMI spec defines the "WMNB" ACPI method to my understanding (in this case in, ATKD namespace). This ACPI method in turn has several methods inside, the most important are Device STatuS and DEVice Set, which these laptops use for almost everything non-standard. These in turn, take a device ID (IIA0) which is a proprietary dword number assigned to some functionality with, sometimes, arguments (IIA1..X). Searching for device ID typically gets you two matches DSTS and DEVS. Calling such WMI methods is already implemented in the driver.

My understanding is that these are more persistent and do not change that much between models (different from the raw ACPI methods).

Simplest and safest solution for poking these WMI calls is to use the debugfs interface of the `asus-wmi` driver (see <https://github.com/hackbnw/faustus/blob/master/src/faustus.c#L2808>). You can call the interface providing device ID and argument as follows

```
$ echo '0x00110018' | sudo tee /sys/kernel/debug/faustus/dev_id 
0x00110018
$ echo '0x00000001' | sudo tee /sys/kernel/debug/faustus/ctrl_param
0x00000001
$ sudo cat /sys/kernel/debug/faustus/devs
DEVS(0x110018, 0x1) = 0x1
$ sudo cat /sys/kernel/debug/faustus/dsts
DSTS(0x110018) = 0x10003
```

(the `faustus` part of the path is `asus-nb-wmi` on mainline, the example would switch the fan boost mode on FX series to overboost, you'll have to change dev ID and argument).

Here are some general links too:

- https://wiki.ubuntu.com/Kernel/Reference/WMI
- https://lwn.net/Articles/391230/

hackbnw, 2021-04-20:
I'm fairly certain though that the fan boost mode is _QDD, but it isn't likely that this is how it was supposed to be called. This need further research.

dsm-iv-tr, 2021-04-20:
Thank you for the advice!

I've been looking around the DSDT armed with what you told me. Here's what I think.

`SFSS`, `SFBM`, and `SFNM` could be "Set Fan Silent S____", "Set Fan Boost Mode", and "Set Fan Normal Mode".

`SFSS` and `SFBM` seem to take one argument - `IIA1`.

`SFNM` is empty, but I did find `SFNH`, which is:

```
Method (SFNH, 1, NotSerialized)
        {
            If (\_SB.PCI0.LPCB.EC0.ECOK)
            {
                Acquire (\_SB.PCI0.LPCB.EC0.MUT0, 0x07D0)
                Store (One, \_SB.PCI0.LPCB.EC0.XFFG)
                If (LEqual (Arg0, Zero))
                {
                    Store (One, \_SB.PCI0.LPCB.EC0.FNED)
                }

                If (LEqual (Arg0, One))
                {
                    Store (Zero, \_SB.PCI0.LPCB.EC0.FNED)
                }

                Release (\_SB.PCI0.LPCB.EC0.MUT0)
            }
        }
```

which appears to toggle between setting `1` and `0` for this location depending on the value of `Arg0`.

`SFNH` is not referenced anywhere else in the DSDT, though, so I don't know what value it will be to us.

dsm-iv-tr, 2021-04-20:
Here is a log of my experimentation.

```
root@Laptop:/sys/kernel/debug/asus-nb-wmi# echo '0x00110015' > dev_id 
root@Laptop:/sys/kernel/debug/asus-nb-wmi# echo '0x00000000' > ctrl_param
root@Laptop:/sys/kernel/debug/asus-nb-wmi# cat devs
DEVS(0x110015, 0x0) = 0x1
root@Laptop:/sys/kernel/debug/asus-nb-wmi# cat dsts
cat: dsts: No such device
```

The first part makes sense to me, because I expect to get 0x01 back:

```
                    If (LEqual (IIA0, 0x00110015))
                    {
                        SFSS (IIA1)
                        Return (One)
                    }
```

I don't know why `dsts` doesn't exist, though.

dsm-iv-tr, 2021-04-26:
Looking around, I found this method to trigger fans: https://www.gabsoftware.com/tutorials/tutorial-how-i-forced-my-laptop-fan-to-spin-under-linux-fujitsu-amilo-xi-1554-1526-1546/

Essentially, the author fakes the temperature via a small added function inside their DSDT (where the kernel loads this modified table), then uses an external tool to control the execution of that ACPI call based on a temperature. So, for example, `sensors` or `thermald` might be used to control the override function.

Obviously a hack, but an interesting idea.

dsm-iv-tr, 2021-04-26:
I also found this while running

`echo '1' > /sys/class/thermal/cooling_device0/cur_state`

```
[  380.182701] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.LPCB.HEC.ECAV], AE_NOT_FOUND (20190816/psargs-330)
[  380.182738] 
               Initialized Local Variables for Method [FNCL]:
[  380.182741]   Local0: 00000000efd336f9 <Obj>           Integer 0000000000000001
[  380.182755]   Local1: 000000007b9a308c <Obj>           Integer 0000000000000000
[  380.182766]   Local2: 000000003e419e9e <Obj>           Integer 0000000000000000
[  380.182776]   Local3: 000000002eb1a808 <Obj>           Integer 0000000000000000
[  380.182784]   Local4: 000000008f2045d6 <Obj>           Integer 0000000000000000
[  380.182792]   Local5: 00000000f6d10c67 <Obj>           Integer 0000000000000000
[  380.182801]   Local6: 00000000e5174b7e <Obj>           Integer 0000000000000000
[  380.182812] No Arguments are initialized for method [FNCL]
[  380.182819] ACPI Error: Aborting method \_TZ.FNCL due to previous error (AE_NOT_FOUND) (20190816/psparse-529)
[  380.182846] ACPI Error: Aborting method \_TZ.FN00._ON due to previous error (AE_NOT_FOUND) (20190816/psparse-529)
```

Could be a hint to the correct methods?

dsm-iv-tr, 2021-05-17:
@hackbnw - have you made any headway with this? I have some time to fool around if you have any suggestions.

hackbnw, 2021-05-17:
Hi, sorry, haven't yet had time to look into it. Possibly this week.

hackbnw, 2021-05-26:
Hi @dsm-iv-tr thanks for you research, I see you tried a lot. I have no idea what SFSS does, it does calculate something. Thinkng about it, it might in fact be some measurement method, that takes into account the boost mode or something. The other error with AE_NOT_FOUND is not likely to be related.

Wondering, could you please try the following:

```
echo '0x00100021' | sudo tee /sys/kernel/debug/faustus/dev_id
echo '0x99' | sudo tee /sys/kernel/debug/faustus/ctrl_param
sudo cat /sys/kernel/debug/faustus/devs
```

This just emulates fan key on mine. Might it be it switches the boost mode too on yours?

Slowly running of ideas.. the _QDD must be it, but it is not called anywhere, the remaining options would be it's called from somewhere hidden in WMI blobs (?), the decompiler has errors and missing something, or it's in some other ACPI tables. DSDT is only one of the tables, but typically most interesting, one could try dumping and decompiling the rest of them, but take great discretion when posting them on the internet, some have serial numbers / product keys there, e.g. MSDM. I guess the call might be in some of SSDT tables, I haven't looked in mine yet, but this seems a bit far fetched. No idea. One could definitely also try calling the ACPI _QDD method directly to verify that it's at least right functionality, I will be looking into it further. This is strange.

dsm-iv-tr, 2021-05-27:
I tried this with the `asus-nb-wmi` module, this is the result.

```
root@Laptop:~# echo '0x00100021' | sudo tee /sys/kernel/debug/asus-nb-wmi/dev_id 
0x00100021
root@Laptop:~# echo '0x99' | sudo tee /sys/kernel/debug/asus-nb-wmi/ctrl_param
0x99
root@Laptop:~# sudo cat /sys/kernel/debug/as
asoc/        asus-nb-wmi/ 
root@Laptop:~# sudo cat /sys/kernel/debug/asus-nb-wmi/devs 
DEVS(0x100021, 0x99) = 0x0
```

`dmesg` says:

```
[ 1643.687320] asus_wmi: Unknown key ae pressed
```

... and nothing else happens. It does appear as if it's just emulating the keypress.

hackbnw, 2021-05-27:
Ok, thank you for testing. Was worth trying. I'll be looking further.

hackbnw, 2021-05-27:
Ok, the _QDD is the queued event handler (event handler for an event generated by EC):

https://wiki.ubuntu.com/Kernel/Reference/ACPITricksAndTips

https://uefi.org/specs/ACPI/6.4/05_ACPI_Software_Programming_Model/ACPI_Software_Programming_Model.html#exx-lxx-and-qxx-methods-for-gpe-processing

dsm-iv-tr, 2021-05-27:
This is a great find!

I can't help being a little amused and wondering what ASUS was thinking with my laptop, lol.

I can dump and decompile the rest of my machine's tables. I'll report back when I do.

I am confused partly because I am not familiar with the ACPI spec, but also because it makes no sense that ASUS would radically change such fundamental things per-model. I don't understand that kind of thinking, if that's what has happened.

dsm-iv-tr, 2021-06-23:
I found some time to dump and extract the rest of my tables.

[Here's a zipfile of them.](https://github.com/hackbnw/faustus/files/6704257/tables.zip)

I had a look through, I don't think there's anything super-confidential or related to serial numbers/HWIDs, but even if there is, it's no bother to me.

Will these be of any help to you?

dsm-iv-tr, 2021-08-17:
I had a few minutes to play with this again today.

I tried to scan the SMBus for anything interesting:

```
$ sudo i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- 08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- 35 36 -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- 52 -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                        
```

 I found that on Bus 0 @ 0x44, there is an interesting set of further registers:

```
$ sudo i2cdump -y 0 0x44
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 00 00 04 00 00 00 00 00 30 13 22 03 17 08 21    ...?.....0?"???!
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
```

In particular, 0x09 increases once per second. 0x0a increases with every minute - so a clock of some kind.

I will poke around more later.

gh-darvishani, 2022-01-17:
i am faced with this issue
my laptop is fx506li too.
here is my config 
`inxi -Fxz
System:    Kernel: 5.10.0-1055-oem x86_64 bits: 64 compiler: N/A Console: tty 3 Distro: Ubuntu 20.04.3 LTS (Focal Fossa) 
Machine:   Type: Laptop System: ASUSTeK product: ASUS TUF Gaming F15 FX506LI_FX506LI v: 1.0 serial: <filter> 
           Mobo: ASUSTeK model: FX506LI v: 1.0 serial: <filter> UEFI: American Megatrends v: FX506LI.310 date: 11/26/2021 
Battery:   ID-1: BAT1 charge: 40.4 Wh condition: 40.6/48.2 Wh (84%) model: ASUS A32-K55 status: Unknown 
CPU:       Topology: 8-Core model: Intel Core i7-10870H bits: 64 type: MT MCP arch: N/A L2 cache: 16.0 MiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 70399 
           Speed: 859 MHz min/max: 800/5000 MHz Core speeds (MHz): 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800 7: 800 8: 800 
           9: 800 10: 800 11: 800 12: 800 13: 800 14: 800 15: 800 16: 800 
Graphics:  Device-1: Intel UHD Graphics vendor: ASUSTeK driver: i915 v: kernel bus ID: 00:02.0 
           Device-2: NVIDIA vendor: ASUSTeK driver: N/A bus ID: 01:00.0 
           Display: server: X.Org 1.20.13 driver: modesetting unloaded: fbdev,vesa tty: N/A 
           OpenGL: renderer: Mesa Intel UHD Graphics (CML GT2) v: 4.6 Mesa 21.0.3 direct render: Yes 
Audio:     Device-1: Intel Comet Lake PCH cAVS vendor: ASUSTeK driver: snd_hda_intel v: kernel bus ID: 00:1f.3 
           Device-2: NVIDIA vendor: ASUSTeK driver: snd_hda_intel v: kernel bus ID: 01:00.1 
           Device-3: DCMT USB Condenser Microphone type: USB driver: hid-generic,snd-usb-audio,usbhid bus ID: 1-1:2 
           Sound Server: ALSA v: k5.10.0-1055-oem 
Network:   Device-1: Intel Wi-Fi 6 AX201 driver: iwlwifi v: kernel port: 5000 bus ID: 00:14.3 
           IF: wlo1 state: up mac: <filter> 
           Device-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: ASUSTeK driver: r8169 v: kernel 
           port: 3000 bus ID: 03:00.0 
           IF: enp3s0 state: down mac: <filter> 
           IF-ID-1: docker0 state: down mac: <filter> 
Drives:    Local Storage: total: 715.41 GiB used: 341.92 GiB (47.8%) 
           ID-1: /dev/nvme0n1 vendor: Western Digital model: PC SN530 SDBPNPZ-256G-1002 size: 238.47 GiB 
           ID-2: /dev/sda vendor: Silicon Power model: SPCC Solid State Disk size: 476.94 GiB temp: 40 C 
Partition: ID-1: / size: 298.35 GiB used: 221.55 GiB (74.3%) fs: ext4 dev: /dev/sda1 
Sensors:   System Temperatures: cpu: 68.0 C mobo: 27.8 C 
           Fan Speeds (RPM): N/A 
Info:      Processes: 412 Uptime: 1h 16m Memory: 31.20 GiB used: 3.15 GiB (10.1%) Init: systemd runlevel: 5 Compilers: 
           gcc: 9.3.0 Shell: bash v: 5.0.17 inxi: 3.0.38 
`
as you can see the fans are not detect.
could you find any success. if yes let me know

abhijeetviswa, 2022-01-25:
I have an FX506HC.

```
BIOS Information
        Vendor: American Megatrends International, LLC.
        Version: FX506HC.310
        Release Date: 11/25/2021
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX506HC
        Version: 1.0
```

Writing values to `throttle_thermal_policy` works as expected. It changes the fan speed as reported by `lm-sensors` and was also validated by listening to fan noise. Using the Fn+F5 keyboard combo also works.

Apart from this keyboard backlight brightness as well as colour is controllable by writing to sysfs.

dsm-iv-tr, 2022-06-10:
I just upgraded my kernel to ` 5.14.0-1042-oem` and my BIOS to the `FX506LI.310` release.

With the stock `asus-nb-wmi` module:

- `throttle_thermal_policy` changes values properly (0/1/2)
- `dmesg` no longer contains messages about invalid keycodes
- keyboard backlight brightness adjustment works fine
- fan does not respond to changes in `throttle_thermal_policy`
- no RGB control

With `faustus`:
- `throttle_thermal_policy` changes values properly (0/1/2)
- `dmesg` no longer contains messages about invalid keycodes
- keyboard backlight brightness adjustment works fine
- fan does not respond to changes in `throttle_thermal_policy`
- RGB keyboard control works fine