hackbnw/faustus #64: Set Start-Up Mode

Syzygianinfern0, 2021-04-08:
I would like the fans to be on Turbo Mode automatically at startup. Is there any way I can achieve this? Currently, it starts in the Normal Mode every time.

cromerc, 2021-04-08:
TUF-Manager restores the fan speed on login: https://git.cromer.cl/cromer/tuf-manager

hackbnw, 2021-04-18:
More low-tech you could also add a command to `/etc/rc.local`, which is a file that gets executed on startup or create a systemd unit. The rc.local should be marked writable only by root and executable, and have a shebang line in the beginning (be careful to verify the file syntax is correct, otherwise you could potentially break boot process). This will however only change the mode after system boots, not before, as the setting is not saved by BIOS itself.

closed 2021-04-18 by hackbnw