hackbnw/faustus #65: Working on AsusTUF-705DT with KDE-Neon 5.21

Corrupter-rot, 2021-05-06:
Kernel version- 5.8.0-50-generic

Output for the command: 
```
$ sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
        Vendor: American Megatrends Inc.
        Version: FX705DT.315
        Release Date: 09/22/2020
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX705DT
        Version: 1.0
```

Just installed so far RGB's are working.
But can't change the fans profiles. Looks like the system is managing the fans profile.

hackbnw, 2021-06-02:
Hello, thanks for feedback.

The fans comment  is interesting. Do you mean you can't switch the modes or you can't modify the specific temperature profiles of the modes? (I've seen another project on github, which did sometimes allow that, want to verify that my understanding is correct). Can you please check output of `dmesg` as you are switching the modes? Is the problem occurs when writing to sysfs or also with the Fn-F5 shortcut? Up until now the problem with mode switching was present only with FXxxxII/LI/IU models. If that is the case it might seem to suggest that the behavior is broken in all recent BIOSes, as there was successful report about fan mode switching for this exact model, but with 308 BIOS (see https://github.com/hackbnw/faustus/issues/37). Could you please provide dump of your DSDT table (if this is the case) as shown [here](https://github.com/hackbnw/faustus#information-to-include-in-feedback) for comparison? We are currently trying to understand the issue there.

Regarding your question about keyboard backlight, I don't remember exactly, but I think you can disable it in sleep mode by setting the [kbbl_flags](https://github.com/hackbnw/faustus#rgb-backlight) in the script to a bitmask which does not include the sleep mode (that would be 0a).

dsm-iv-tr, 2021-06-23:
> It's a request:
> when laptops goes to sleep the backlight goes into breathing mode even if they are turned off. It was the same in windows but there was a option to turn it off. If you can provide the option in this too that would be helpful.

Testing, @hackbnw's suggestion works fine, simply OR the values you want together. 08 | 02 = 0A, so

```
echo 0a > /sys/devices/platform/faustus/kbbl/kbbl_flags
```

Corrupter-rot, 2021-06-27:
Recently I did a clean install of my distro. After that did some digging and found some things.

1. After linux 5.7 kernel version the `Fn+f5` function works natively on my laptop. It did change the fan profile but it didn't give any feedback so i used [leonmaxx/asus_fanmode](https://github.com/leonmaxx/asus_fanmode) which automatically managed the fan profile depending on the Temps.
2. As for the breathing effect i found that after i put the lid down when the laptop is in sleep the lights are off. Thanks for the solution @dsm-iv-tr @hackbnw  . I will try after installing the program again.

After posting this comment I am installing the program again and let you know if thing are working as expected.

closed 2022-07-23 by Corrupter-rot