hackbnw/faustus #67: Working on ASUS F17 FX706Li

alkinoy, 2021-05-20:
$ sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FX706LI.307
	Release Date: 12/17/2020
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FX706LI
	Version: 1.0

hackbnw, 2021-06-02:
Hello, thanks for the feedback, can you please specify what exactly is working? I guess RGB backlight control?

Specifically could you please verify, if you can switch fan modes with Fn-F5 and whether do you have aura keys on Fn-Left / Fn-Right (both are known issues on FX506LI)? Also, if you could provide DSDT table for comparison (as shown [here](https://github.com/hackbnw/faustus/#information-to-include-in-feedback)) that might be helpful for finding the solution.

alkinoy, 2021-06-10:
morning.
I had a brand new laptop and found that I can't change keyboard backlight color with newly installed Ubuntu 20.04. It continuously change colors. Control of brightness worked well, but color didn't. I've tried this utilite and now I have constant green light (as desired). Probably I didn't install it in proper way because when I tried to change color (just for test) I got error:
`sudo ./set_rgb.sh
./set_rgb.sh: line 4: /sys/devices/platform/faustus/kbbl/kbbl_red: No such file or directory
./set_rgb.sh: line 6: /sys/devices/platform/faustus/kbbl/kbbl_green: No such file or directory
./set_rgb.sh: line 8: /sys/devices/platform/faustus/kbbl/kbbl_blue: No such file or directory
./set_rgb.sh: line 10: /sys/devices/platform/faustus/kbbl/kbbl_mode: No such file or directory
./set_rgb.sh: line 12: /sys/devices/platform/faustus/kbbl/kbbl_speed: No such file or directory
./set_rgb.sh: line 14: /sys/devices/platform/faustus/kbbl/kbbl_flags: No such file or directory
./set_rgb.sh: line 16: /sys/devices/platform/faustus/kbbl/kbbl_set: No such file or directory
`
I didn't use fan control because it seems to be ok - it change fan speed according to load.

Sweetchuck, 2021-07-27:
I can confirm that it works on ASUS TUF FX706LI, but the `openrgb` and/or the `set_rgb.sh` has to be run as "root" with `sudo`

I had to use the `sudo insmod src/faustus.ko let_it_burn=1` command.