hackbnw/faustus #70: Working in Asus Tuf A17 FA706QM - Ubuntu 20.04

JuanuSt, 2021-06-19:
Ubuntu 20.04 - kernel 5.12.12-051212-generic

BIOS Information
	Vendor: American Megatrends International, LLC.
	Version: FA706QM.307
	Release Date: 06/02/2021
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA706QM
	Version: 1.0

mrtglez, 2021-07-14:
I tried it but nop

JuanuSt, 2021-08-05:
I did several things during the installation of Ubuntu, I had problems with wifi and bluetooth combo adapter and of course with Nvidia propietary drivers, so it could just have been a lucky combination of events.

I took few notes during the setup, all I can do is to share them:

**0.- Update BIOS firmware**
https://www.asus.com/support/FAQ/1012815/
https://www.asus.com/supportonly/FA706QM/HelpDesk_BIOS/
(New installed version FA706QMAS.307)

**1.- Download Mini.iso and install** (Normal image fails due to UEFI boot)
http://archive.ubuntu.com/ubuntu/dists/focal/main/installer-amd64/current/legacy-images/netboot/mini.iso
(and follow wizard. Very slow step, I chose Desktop minimal and OpenSSH server)

**2.- Update kernel to last stable version.** Use this method because problems in libc version
sudo add-apt-repository ppa:tuxinvader/lts-mainline
sudo apt install linux-image-unsigned-5.12.4-051204-generic linux-modules-5.12.4-051204-generic linux-headers-5.12.4-051204-generic

-- REBOOT --

**3 and 4 are for Nvidia and Wifi/Bluetooth steps** (not shown).

**5.- Keyboard backlight config**. Unofficial driver. Install temporally
(Follow instructions in https://github.com/hackbnw/faustus)

- Blaclist asus drivers for after reboot
nano /etc/modprobe.d/faustus.conf
> blacklist asus_wmi
   blacklist asus_nb_wmi

- Remove asus modules for current test
sudo rmmod asus_nb_wmi
sudo rmmod asus_wmi
dmesg # Verify they are removed

- Clone hack driver
git clone https://github.com/hackbnw/faustus
cd faustus
make

- Test
sudo modprobe sparse-keymap
sudo modprobe wmi
sudo modprobe video
sudo insmod src/faustus.ko
ERROR: could not insert module src/faustus.ko: No such device
sudo insmod ./src/faustus.ko let_it_burn=1 # OK
./set_rgb.sh
edit color and other variables. Test them with this script.

- Current color ubuntu purple
echo 77 > /sys/devices/platform/faustus/kbbl/kbbl_red
echo 21 > /sys/devices/platform/faustus/kbbl/kbbl_green
echo 6f > /sys/devices/platform/faustus/kbbl/kbbl_blue

- Mode: 0 - static color, 1 - breathe, 2 - color cycle, 3 - strobe
echo 0 > /sys/devices/platform/faustus/kbbl/kbbl_mode

- Speed for modes 1 and 2: 0 - slow, 1 - medium, 2 - fast
echo 0 > /sys/devices/platform/faustus/kbbl/kbbl_speed

- Enable: 02 - on boot (before module load) | 08 - awake | 20 - sleep (2a or ff to set all)
echo 2a > /sys/devices/platform/faustus/kbbl/kbbl_flags

- Save: 1 - permanently, 2 - temporarily (reset after reboot)
echo 1 > /sys/devices/platform/faustus/kbbl/kbbl_set

- Install and make it on boot
make dkms
sudo make onboot

-- REBOOT --

After changes and reboot we back to normal drivers, because some hot-keys stop to functions, and the parameters set with experimental driver remain.

- Unblacklist asus drivers
rm /etc/modprobe.d/faustus.conf

- Disable from boot
cd ~/Dowloads/faustus
sudo make noboot

 - Remove faustus
sudo dkms remove faustus/0.2 --all

-- REBOOT --

Good luck