hackbnw/faustus #72: Working on FA506

roketscyntist, 2021-07-22:
Working on my FA506 with let_it_burn:

BIOS Information
        Vendor: American Megatrends Inc.
        Version: FA506IU.316
        Release Date: 03/12/2021
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FA506IU
        Version: 1.0

Only tested the RGB portion using openrgb.

Thanks for this, one less reason to boot into windows!

ic3man5, 2021-09-13:
Can confirm everything is working also. All LED modes and fan modes.

```
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA506IU.314
	Release Date: 12/17/2020
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 5.16
	Firmware Revision: 3.12

```

```
$ ls /sys/devices/platform/faustus/
cpufv  driver  driver_override  fan_boost_mode  hwmon  input  kbbl  leds  modalias  power  subsystem  throttle_thermal_policy  uevent
```

FN+F5 changes `throttle_thermal_policy` but does not change `fan_boost_mode`.

ic3man5, 2021-09-13:
I'm also on Fedora 34 and confirmed it works with openrgb also.