hackbnw/faustus #73: Working on FX506HM but settings don't persist after reboot

lovesh, 2021-07-24:
Works on ASUS TUF F15 FX506HM_FX566HM when using `let_it_burn=1`. 

OS: Kubuntu 21.10
Kernel: 5.13.4 and 5.14.0-rc2

```
BIOS Information
        Vendor: American Megatrends International, LLC.
        Version: FX506HM.303
        Release Date: 07/06/2021
--
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: FX506HM
        Version: 1.0
```

[dsdt.aml.zip](https://github.com/hackbnw/faustus/files/6872821/dsdt.aml.zip)

Tried RGB and fan mode. Settings don't persist on reboot though.

I have an unrelated (to this repo) question. My touchpad isn't detected. Tried many things, details [here](https://askubuntu.com/questions/1352620/touchpad-not-working-on-asus-tuf-f15-2021-edition). Any help would be appreciated.

cromerc, 2021-07-24:
As far as the settings not persisting, that is known and normal. If you want persistent options you can try TUF-Manager: https://git.cromer.cl/cromer/tuf-manager

TUF-Manager runs in the background as a service and remembers your color settings and fan speed. Then the next time you login and it runs it will restore those settings. Plus various other features such as controlling the colors and fan speed, notifications to show what the current fan speed is, a gui interface, a cli interface(for the purist that don't want a gui or want to change it from a terminal), etc.

Andy3153, 2021-08-15:
Can confirm. It works flawlessly on FX506HM, running on Arch Linux