hackbnw/faustus #79: ASUS TUF FX505DT: Fan control does not work

UtkarshVerma, 2021-09-01:
I can't get fan control to work on my laptop. I do get the sysfs file `throttle_thermal_policy` and its value changes when I press the fan speed button, but the fan doesn't slow down. My CPU fan always runs at high speed.

It used to work for me before, but I don't know why it doesn't work now.

I am using a custom compiled kernel (v5.13.12) on Void Linux. I think I have the necessary options checked in the kernel config, but could you please mention once what needs to be enabled in the kernel to get fan speed control to work, if the kernel is the issue here.

cromerc, 2021-10-08:
> I have a similar laptop, ASUS TUF Gaming FX505DT_FX505DT. Are there instructions on how to get the keyboard to change color in linux?

There are instructions in the README of the repo:
https://github.com/hackbnw/faustus#rgb-backlight

Or if you want a GUI to do it there is TUF Manager: https://git.cromer.cl/cromer/tuf-manager

nickcat325, 2021-10-09:
> There are instructions in the README of the repo: https://github.com/hackbnw/faustus#rgb-backlight
>
> Or if you want a GUI to do it there is TUF Manager: https://git.cromer.cl/cromer/tuf-manager

Thank you, it worked.