hackbnw/faustus #87: Not working on Asus FX506L

TheArchKnight, 2021-12-02:
Whenever i try to load the driver on my Arch Linux installation, i'll get the error message: 
`ERROR: could not insert module src/faustus.ko: No such device`
In theory as pointed in the github information page, it should work without trouble, but it isn't (at least in my device)

shadowelite-sec, 2022-01-13:
> Whenever i try to load the driver on my Arch Linux installation, i'll get the error message: `ERROR: could not insert module src/faustus.ko: No such device` In theory as pointed in the github information page, it should work without trouble, but it isn't (at least in my device)

same issue

bonohub13, 2022-08-16:
Hi! Had the same issue, but was able to solve it by adding `let_it_burn=1` when loading the module manually.
If it works, maybe following the instructions that I added under this issue.
https://github.com/hackbnw/faustus/issues/106