hackbnw/faustus #93: Does ASUS TUF Gaming A15 FA506IC work with the driver?

TanawatJukmongkol, 2022-02-22:
Hi, I've got recommended by CalcProgrammer1 to try out this driver, but I don't want to ruin or possibly burn my new expensive $926 laptop (I'm living in a third world country, and I wanted to get into programming. The current chip shortage doesn't help either. The price is ridiculous).
I use Parrot OS 5.0 (LTS), with Parrot's custom hardened kernel 5.14.0-9parrot1-amd64. Will it work? Thanks!

logicsys, 2022-07-24:
Yes, Ive just tried it on a FA506IC, works great, have finally been able to stop the rainbow colour cycle on the keyboard. No smoke after a few reboots and stress tests either.

TanawatJukmongkol, 2022-07-25:
I can't change the color via OpenRGB or the terminal via the command `cat "ff" > kbbl_blue`. I also can't start up the module, it errors out `modprobe: FATAL: Module faustus not found in directory /lib/modules/5.18.10-76051810-generic`, also the install gives this error:
```
make -C /lib/modules/5.18.10-76051810-generic/build M=/home/popos/Packages/faustus modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.18.10-76051810-generic'
  INSTALL /lib/modules/5.18.10-76051810-generic/extra/src/faustus.ko
  SIGN    /lib/modules/5.18.10-76051810-generic/extra/src/faustus.ko
At main.c:160:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: ../crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: ../crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.18.10-76051810-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.18.10-76051810-generic'
```

logicsys, 2022-07-25:
Looks like you might have secure boot enabled, and then an ssl lib error
occurs when the setup is trying to sign the kernel module.

If you aren't bothered about secure boot, disable it in your bios, then try
installing the faustus kernel module once more

TanawatJukmongkol, 2022-07-26:
> Looks like you might have secure boot enabled, and then an ssl lib error occurs when the setup is trying to sign the kernel module. If you aren't bothered about secure boot, disable it in your bios, then try installing the faustus kernel module once more
> […](#)

Thank you for the reply! I can't disable TPM on my machine, there's no option for it in the UEFI, even though there **is** a TPM 2.0 module. How do I disable it? Thank you!