hackbnw/faustus #103: FX506LI.306

Mjorldir, 2022-06-28:
Hi,
I followed the how to install from Readme.md and this was the result:

`running: sudo rmmod asus_wmi
rmmod: ERROR: Module asus_wmi is in use by: mfd_aaeon

running: 
sudo insmod src/faustus.ko 
sudo insmod ./src/faustus.ko let_it_burn=1  -> in sudo su mode
both answers are the same
insmod: ERROR: could not insert module src/faustus.ko: Operation not permitted

sudo dmidecode | grep "BIOS Inf\|Board Inf" -A 3
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FX506LI.306
	Release Date: 09/11/2020
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FX506LI
	Version: 1.0`

After this I can now longer use Fn+ arrows to have leds change in different colors and I could control speed of blinking.
I have Linux mint 20.3 Cinnamon as OS.

Is there a possibility to have this driver working for my laptop model?

Thank you..!!!

Mjorldir, 2022-06-28:
Solved the error of this command:
sudo insmod src/faustus.ko
by disabling secure Boot on BIOS settings.
Now running command:
dmesg | tail  ->

`[   74.635749] faustus: loading out-of-tree module taints kernel.
[   74.635905] faustus: module verification failed: signature and/or required key missing - tainting kernel
[   74.637615] faustus: DMI check: FX506LI
[   74.638207] faustus: Initialization: 0x1
[   74.638571] faustus: BIOS WMI version: 8.1
[   74.638881] faustus: SFUN value: 0x4a0061
[   74.638887] faustus faustus: Use DSTS
[   74.638891] faustus faustus: Enable event queue
[   74.645882] input: Asus WMI hotkeys as /devices/platform/faustus/input/input27
[   74.658655] ACPI: battery: new extension: ASUS Battery Extension`

2 line error, what can it be?

Mjorldir, 2022-06-28:
With OpenRGB it doesn't work but running `sudo ./set_rgb.sh` works perfectly..!!
Solved

closed 2022-06-28 by Mjorldir

cromerc, 2022-06-28:
> With OpenRGB it doesn't work but running `sudo ./set_rgb.sh` works perfectly..!!
> Solved

Don't know about OpenRGB, but TUF-Manager does work with this driver: https://git.cromer.cl/cromer/tuf-manager