hackbnw/faustus #107: Can't get faustus to run at boot or dkms to add it to new kernels

mfreeman72, 2022-08-19:
My laptop is an ASUS TUF Gaming A17 FA706IH_TUF706IH. I'm running Linux Mint as my OS.

When I bought the laptop in December 2020, I got faustus working on Linux Mint 20.x (based on Ubuntu 20.04), and it has been working flawlessly since then. A week or so ago, I did a fresh install of Linux Mint 21 (based on Ubuntu 22.04). I tried to install faustus again, and this time I can't get it to boot. I don't remember if there was anything I needed to do to from last time, because that was more than 1 1/2 years ago.

I followed the instructions on your github, except that I added let_it_burn=1 to the modprobe. I could then change the light settings. I also followed the dkms and Ubuntu onboot instructions. But when I rebooted, faustus wouldn't automatically load. I had to manually enter sudo modprobe faustus let_it_burn=1 to get it working again.

I then manually edited the /etc/modules-load.d/faustus.conf file to read: faustus let_it_burn=1, which didn't fix the problem. I also found a post of someone saying they made it say options faustus let_it_burn=1, which also didn't fix the problem.

I've also discovered that when I install a new kernel, dkms isn't adding it, and I have to manually recompile it into the kernel each time.

I've redone the installation process following your instructions about 10 times or so, and I'm getting the same result each time. I know faustus works with my system, because I've used it reliably for the last year and a half, and I can manually get it going. It's just the loading automatically at boot and getting dkms adding it to new kernels that's giving me issues ever since I installed the new OS.

Any help would be appreciated! Thanks!

cromerc, 2022-08-19:
If you did it 10 times... you skipped the same step in the instructions 10 times...
https://github.com/hackbnw/faustus#using-dkms

You have to install the dkms module for it to be used by dkms.

bonohub13, 2022-08-19:
> I then manually edited the /etc/modules-load.d/faustus.conf file to read: faustus let_it_burn=1, which didn't fix the problem. I also found a post of someone saying they made it say options faustus let_it_burn=1, which also didn't fix the problem.

You've edited the wrong config file. You're supposed to edit `/etc/modprobe.d/faustus.conf` instead of `/etc/modules-load.d/faustus.conf`.
Try appending `options faustus let_it_burn=1` to `/etc/modprobe.d/faustus.conf` and rebuilding the initramfs.
And remove the line `options faustus let_it_burn=1` from `/etc/modules-load.d/faustus.conf` just in case. ;)

mfreeman72, 2022-08-19:
@cromerc No, I did that step. I partly figured out what's going on with dkms. I'm running the Xanmod kernel, because it solves some issues that the regular Ubuntu-built kernel has with my specific laptop. If I "sudo make dkms" with the Ubuntu-built kernel, it sticks. But if I build it with the Xanmod kernel, it says: "Error! Bad return status for module build on kernel: 5.18.18-xanmod1 (x86_64). In Linux Mint 20.x, I originally built faustus with the Ubuntu kernel, and when I installed the Xanmod kernel, it just transferred over to it through dkms without problems. I don't know why it's not working in Linux Mint 21.

That aside, even with the Ubuntu-built kernel, I have to manually sudo modprobe faustus let_it_burn=1 after every boot. It never runs automatically, like it should.

mfreeman72, 2022-08-19:
@bonohub13 Yeah, I don't know how I made that mistake! Anyway, I put options faustus let_it_burn=1 in /etc/modprobe.d/faustus.conf, and removed it from the other location, and it's still not loading on boot, no matter what kernel I use. I may just wipe every trace of faustus from the system and start again from scratch.

bonohub13, 2022-08-19:
@mfreeman72 Did you rebuild the initramfs?

bonohub13, 2022-08-19:
Try configuring the `/etc/modules-load.d/faustus.conf` and `/etc/modprobe.d/faustus.conf` from [this issue](https://github.com/hackbnw/faustus/issues/106).
The only thing you should have to change is the command necessary to update the initramfs.
Run `sudo update-initramfs -u` instead of `sudo mkinitcpio -P` for Linux Mint.

mfreeman72, 2022-08-20:
@bonohub13 Thanks for the input! I've already wiped faustus from the system and completely uninstalled/purged the Xanmod kernel. I git cloned and reinstalled faustus from scratch, and installed it again, this time making sure to use the right /etc/modprobe.d folder location. That worked. It now runs on the Ubuntu kernel. However, reinstalling the Xanmod kernel still gives me problems. I've run `sudo update-initramfs -u` to make sure it was updated, even though that should have been covered by the reinstall. Still doesn't work on Xanmod.

Typing sudo modprobe faustus gives me:
`modprobe: FATAL: Module faustus not found in directory /lib/modules/5.18.18-xanmod1`

Trying to `sudo make dkms` again gives me: 
```
Error! The module/version combo: faustus-0.1 is not located in the DKMS tree.
Deleting module faustus-0.2 completely from the DKMS tree.
dkms add .
Creating symlink /var/lib/dkms/faustus/0.2/source -> /usr/src/faustus-0.2
dkms install -m faustus -v 0.2

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.18.18-xanmod1 -C /lib/modules/5.18.18-xanmod1/build M=/var/lib/dkms/faustus/0.2/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.18-xanmod1 (x86_64)
Consult /var/lib/dkms/faustus/0.2/build/make.log for more information.
make: *** [Makefile:24: dkms] Error 10
```

Looking at the make.log shows this:
```
DKMS make.log for faustus-0.2 for kernel 5.18.18-xanmod1 (x86_64)
Fri Aug 19 06:01:27 PM PDT 2022
make[1]: Entering directory '/usr/src/linux-headers-5.18.18-xanmod1'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-12) 11.2.0
  You are using:           gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  CC [M]  /var/lib/dkms/faustus/0.2/build/src/faustus.o
In file included from ./include/linux/rhashtable-types.h:15,
                 from ./include/linux/ipc.h:7,
                 from ./include/uapi/linux/sem.h:5,
                 from ./include/linux/sem.h:5,
                 from ./include/linux/sched.h:15,
                 from ./include/linux/mm_types.h:6,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from /var/lib/dkms/faustus/0.2/build/src/faustus.c:30:
./include/linux/workqueue.h:152:1: error: expected specifier-qualifier-list before ‘static’
  152 | static inline bool schedule_delayed_work(struct delayed_work *dwork,
      | ^~~~~~
./include/linux/workqueue.h:201:1: warning: null character(s) ignored
  201 |                                                                                                                                                        >
      | ^
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/faustus/0.2/build/src/faustus.o] Error 1
make[1]: *** [Makefile:1845: /var/lib/dkms/faustus/0.2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.18-xanmod1'
```

The only things I can figure out from that is that there's some "expected specifier-qualifier-list before 'static'" error in the include process, but I'm not sure what it means exactly. So I'm not sure where to go from here.

bonohub13, 2022-08-20:
It's only a speculation so I want this to be only taken with a grain of salt, but it could be the gcc version. The Xanmod kernel was built using gcc-11 and it seems to me that the default kernel in Linux Mint isn't. It could be it, but I'm not so sure.
I'm gonna try to dig into this a bit more tho.

mfreeman72, 2022-09-22:
This issue seems to have resolved itself. Not sure how, exactly, but it is now working no matter which kernel I use. Thank you for your help!

closed 2022-09-22 by mfreeman72