hackbnw/faustus #108: Doesn't work on Asus TUF A17 FA706IC

0xNoor, 2022-08-21:
I've tested this model personally and nothing works. 
BIOS:  FA706IC.301
`ERROR: could not insert module src/faustus.ko: No such `device``
You can add this model to the list. If any other information is required, I'll be glad to provide.

bonohub13, 2022-08-21:
Try adding `let_it_burn=1` when loading the module manually.
eg: `sudo insmod src/faustus.ko let_it_burn=1`

0xNoor, 2022-08-22:
I tried using this, and almost everything works. Sorry I didn't read the full documentation at first. 

Fans mode key works (Fn-F5)  :white_check_mark: 
Keyboard Backlit intensity works (Fn-Up, Fn-Down) :white_check_mark: 
Changing RGB keyboard color works using set_rgb.sh and Openrgb  :white_check_mark: 
RGB hot keys (Fn-Left, Fn-Right) doesn't work  :negative_squared_cross_mark: 
 
OS - Kernel info
```
BIOS Information
	Vendor: American Megatrends Inc.
	Version: FA706IC.301
	Release Date: 12/15/2021
--
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: FA706IC
	Version: 1.0

```

Attaching dsdt.aml as tar 
[dsdt.aml.tar.gz](https://github.com/hackbnw/faustus/files/9393437/dsdt.aml.tar.gz)