hackbnw/faustus #115: Build fails on kernel version 6.1.1-1-MANJARO - no acpi_video_set_dmi_backlight_type

S-trace, 2022-12-25:
Hello.
I'm getting a following error when building faustus on kernel version 6.1.1-1-MANJARO.
Build was successful on previous (6.0) kernel releases like 6.0.14-1-MANJARO
```
LANG=C make
make -C /lib/modules/6.1.1-1-MANJARO/build M=/home/s-trace/src/faustus modules
make[1]: Entering directory '/usr/lib/modules/6.1.1-1-MANJARO/build'
  CC [M]  /home/s-trace/src/faustus/src/faustus.o
/home/s-trace/src/faustus/src/faustus.c: In function 'asus_wmi_add':
/home/s-trace/src/faustus/src/faustus.c:3111:17: error: implicit declaration of function 'acpi_video_set_dmi_backlight_type'; did you mean 'acpi_video_get_backlight_type'? [-Werror=implicit-function-declaration]
 3111 |                 acpi_video_set_dmi_backlight_type(acpi_backlight_vendor);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 acpi_video_get_backlight_type
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /home/s-trace/src/faustus/src/faustus.o] Error 1
make[1]: *** [Makefile:1992: /home/s-trace/src/faustus] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.1.1-1-MANJARO/build'
make: *** [Makefile:9: default] Error 2
```

nickcat325, 2022-12-25:
Same for Arch Linux 6.1.1.

nickcat325, 2022-12-26:
I recommend enabling the modules for limited keyboard. This is a temporary solution until someone can find a fix.
/etc/modprobe.d/faustus.conf
```
#blacklist asus_wmi
#blacklist asus_nb_wmi
```

bonohub13, 2023-01-08:
Currently on kernel 6.1.3 running on FA506QM and was able to get faustus working with the following change
``` c
3378   │ #if LINUX_VERSION_CODE < KERNEL_VERSION(6,1,1)
3379   │     /* Some Asus desktop boards export an acpi-video backlight interface,
3380   │        stop this from showing up */
3381   │     chassis_type = dmi_get_system_info(DMI_CHASSIS_TYPE);
3382   │     if (chassis_type && !strcmp(chassis_type, "3"))
3383   │         acpi_video_set_dmi_backlight_type(acpi_backlight_vendor);
3384   │ 
3385   │     if (asus->driver->quirks->wmi_backlight_power)
3386   │         acpi_video_set_dmi_backlight_type(acpi_backlight_vendor);
3387   │ 
3388   │     if (asus->driver->quirks->wmi_backlight_native)
3389   │         acpi_video_set_dmi_backlight_type(acpi_backlight_native);
3390   │ #endif
```

However I cannot ensure this works on all ASUS devices.
Please use at your own risk.

flintstoney, 2023-01-24:
> I recommend enabling the modules for limited keyboard control. This is a temporary solution until someone can find a fix. /etc/modprobe.d/faustus.conf
> 
> ```
> #blacklist asus_wmi
> #blacklist asus_nb_wmi
> ```

I downgraded the kernel and the linux-headers to 6.0.12 and it works fine for now until it will bw fixed

Kartikeya-IYER, 2023-02-01:
According to [a comment](https://github.com/hackbnw/faustus/issues/47#issuecomment-1356151280) on stickied issue #47, faustus may no longer be required on kernel versions >= 6.1. For those distros which roll their own kernels instead of using the vanilla kernel, it may take a while before the distro-maintainer version of kernel 6.1 is made available.

For those already on kernel version >= 6.1, I guess that it might be worth trying the asus driver version in the vanilla kernel instead of faustus compiled from this repo...