hackbnw/faustus #118: fiailed to compile with kernel 6.1.7

flintstoney, 2023-01-24:
it's ending with this error during the compile. 

```
DKMS make.log for faustus-0.2 for kernel 6.1.7-arch1-1 (x86_64)
Tue Jan 24 12:52:08 PM EET 2023
make[1]: Entering directory '/usr/lib/modules/6.1.7-arch1-1/build'
  CC [M]  /var/lib/dkms/faustus/0.2/build/src/faustus.o
/var/lib/dkms/faustus/0.2/build/src/faustus.c: In function ‘asus_wmi_add’:
/var/lib/dkms/faustus/0.2/build/src/faustus.c:3111:17: error: implicit declaration of function ‘acpi_video_set_dmi_backlight_type’; did you mean ‘acpi_video_get_backlight_type’? [-Werror=implicit-function-declaration]
 3111 |                 acpi_video_set_dmi_backlight_type(acpi_backlight_vendor);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 acpi_video_get_backlight_type
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/faustus/0.2/build/src/faustus.o] Error 1
make[1]: *** [Makefile:1992: /var/lib/dkms/faustus/0.2/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.1.7-arch1-1/build'
```

Kartikeya-IYER, 2023-02-01:
According to [a comment](https://github.com/hackbnw/faustus/issues/47#issuecomment-1356151280) on stickied issue #47, faustus may no longer be required on kernel versions >= 6.1. For those distros which roll their own kernels instead of using the vanilla kernel, it may take a while before the distro-maintainer version of kernel 6.1 is made available.

For those already on kernel version >= 6.1, I guess that it might be worth trying the asus driver version in the kernel instead of faustus compiled from this repo...

mainbord, 2023-05-22:
I compiled faustus with manjaro 6.3.3-1-MANJARO and rgb control works on fx506iu. I did somechanges here
https://github.com/hackbnw/faustus/compare/master...mainbord:faustus:master